slic3r/Slic3r #3316: Feature request:  add solid infill layers under support interface

VanessaE, 2016-05-03:
Consider the idea of printing with a dual-extruder setup - PLA, ABS or some other standard plastic in one extruder, and a soluble material in the other.

Currently one can configure Slic3r to print the object, the majority of the support material, and support-to-object interface material with different extruders.  In an ideal world, one might print PLA for the support and object, and PVA for the interface layers.

As far as I can tell, interface layers are laid down directly on top of sparse support, which would normally work fine if everything is printed with the same material.  However, if the interface layer is being printed with PVA, this sometimes doesn't work too well - the PVA doesn't stick well to the underlying sparse support, at least not if PLA is being used for that.  I imagine HIPS has similar problems.

The best way I can see to fix this is that the user should be able to direct Slic3r to finish the top of the sparse support with one or more 100% solid infill layers in the same material.  On top of those, one or more 100% solid infill layers printed with the support-to-object interface material, then the regular interface layers on top of those, and finally the printed object at the top of the stack.

This image should explain it:
[image]
"Please excuse the crudity of this model. I didn't have time to build it to scale or paint it."  :smile:

lordofhyphens, 2016-05-17:
You can specify the interface layer spacing. Set the interface layer count to 1 and make the interface itself solid.
[image]

VanessaE, 2016-05-17:
That doesn't do what I'm talking about. Rather, it results in this (screenshot of Pronterface because it uses different colors per because it assigns colors per-extruder):

[image]

With two extruders, one can load a regular plastic (i.e. PLA or ABS, red in the image) into the first extruder, and a soluble material (PVA/HIPS, purple) into the second, and have Slic3r only use the second extruder for printing those interface layers.  Problem is, that's dense PVA or HIPS being laid down directly onto sparsely-filled PLA or ABS. There's just not enough underlying plastic for the PVA/HIPS to stick to.   That problem is compounded by Slic3r's tendency to start/stop these interface layers in the middle of open space.

There needs to be an option to top the PLA/ABS with one or more almost-solid layers before putting PVA/HIPS down onto it.

lordofhyphens, 2016-05-17:
Try to hack something into source and then link it to me, I can finish the
rest of the plumbing.

VanessaE, 2016-05-18:
Wish I could - I don't know enough about C++ or perl, let alone Slic3r's codebase, to even begin. :neutral_face:

bubnikv, 2016-10-26:
We at Prusa Research are working on multi-material printing.
http://prusaprinters.org/original-prusa-i3-mk2-multi-material-upgrade-release/

I am sure we will have to face the same issues. I am currently working on supports, I hope I will have something in around two weeks.