slic3r/Slic3r #3317: Slic3r crash with STL file

sebastien, 2016-05-03:
### Version

1.2.9

### Operating system type + version

Ubuntu Linux 16.04LTS 64b

### Behavior

Loading the attached STL file (`crash.stl`) makes slic3r crash with the following backtrace:

```
*** Error in `slic3r': realloc(): invalid old size: 0x0000000004cbe400 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7f480753c725]
/lib/x86_64-linux-gnu/libc.so.6(+0x82bfa)[0x7f4807547bfa]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x179)[0x7f4807548db9]
/home/sebastien/Local/share/Slic3r/lib/vrt/bbbcb3e4bbe8939428f72f745f596360/XS.so(stl_add_facet+0x16b)[0x7f4805b2046b]
/home/sebastien/Local/share/Slic3r/lib/vrt/bbbcb3e4bbe8939428f72f745f596360/XS.so(stl_fill_holes+0x3f9)[0x7f4805b208a9]
/home/sebastien/Local/share/Slic3r/lib/vrt/bbbcb3e4bbe8939428f72f745f596360/XS.so(_ZN6Slic3r12TriangleMesh6repairEv+0xe9)[0x7f4805ab6e49]
/home/sebastien/Local/share/Slic3r/lib/vrt/bbbcb3e4bbe8939428f72f745f596360/XS.so(+0xbbbab)[0x7f4805955bab]
/home/sebastien/Local/share/Slic3r/bin/../dll/libperl.so(Perl_pp_entersub+0x5e3)[0x7f48086b9db3]
/home/sebastien/Local/share/Slic3r/bin/../dll/libperl.so(Perl_runops_standard+0xe)[0x7f48086b83fe]
/home/sebastien/Local/share/Slic3r/bin/../dll/libperl.so(Perl_call_sv+0x48b)[0x7f480865228b]
/home/sebastien/Local/share/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so(_ZN18wxPliEventCallback7HandlerER7wxEvent+0x31c)[0x7f480413f58c]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler23ProcessEventIfMatchesIdERK21wxEventTableEntryBasePS_R7wxEvent+0x50)[0x7f480301bdf0]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler23SearchDynamicEventTableER7wxEvent+0x59)[0x7f480301be79]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler11TryHereOnlyER7wxEvent+0x28)[0x7f480301bff8]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler19ProcessEventLocallyER7wxEvent+0x3d)[0x7f480301c91d]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler12ProcessEventER7wxEvent+0x34)[0x7f480301c974]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3(_ZN12wxWindowBase8TryAfterER7wxEvent+0x7a)[0x7f48036ff8da]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler12ProcessEventER7wxEvent+0xb7)[0x7f480301c9f7]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN12wxEvtHandler18SafelyProcessEventER7wxEvent+0x14)[0x7f480301c0c4]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3(_ZNK12wxWindowBase17HandleWindowEventER7wxEvent+0x10)[0x7f48036fedf0]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3(_ZN13wxToolBarBase11OnLeftClickEib+0x55)[0x7f48036e31a5]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x101d4)[0x7f48017411d4]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xc06)[0x7f480175b9a6]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_by_name+0x4a5)[0x7f480175c555]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x101d4)[0x7f48017411d4]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xc06)[0x7f480175b9a6]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7f480175c08f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x8cf35)[0x7f4802879f35]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145)[0x7f4801740fa5]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x21afc)[0x7f4801752afc]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xfbc)[0x7f480175bd5c]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7f480175c08f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x8be79)[0x7f4802878e79]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x132afc)[0x7f480291fafc]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x145)[0x7f4801740fa5]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x2256e)[0x7f480175356e]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xa59)[0x7f480175b7f9]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7f480175c08f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x24a8cc)[0x7f4802a378cc]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_propagate_event+0xc4)[0x7f480291e294]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main_do_event+0x2cb)[0x7f480291e64b]
/usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0(+0x5ac4c)[0x7f4802592c4c]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7)[0x7f4801064127]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4a380)[0x7f4801064380]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc2)[0x7f48010646a2]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main+0xb7)[0x7f480291d6a7]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_gtk2u_core-2.9.so.3(_ZN14wxGUIEventLoop3RunEv+0x4b)[0x7f480351ef7b]
/home/sebastien/Local/share/Slic3r/bin/../dll/libwx_baseu-2.9.so.3(_ZN16wxAppConsoleBase8MainLoopEv+0x46)[0x7f4802ea4ca6]
/home/sebastien/Local/share/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so(_ZN8wxPliApp8MainLoopEv+0x31)[0x7f4804054031]
/home/sebastien/Local/share/Slic3r/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so(+0xbc3c2)[0x7f48040123c2]
/home/sebastien/Local/share/Slic3r/bin/../dll/libperl.so(Perl_pp_entersub+0x5e3)[0x7f48086b9db3]
/home/sebastien/Local/share/Slic3r/bin/../dll/libperl.so(Perl_runops_standard+0xe)[0x7f48086b83fe]
/home/sebastien/Local/share/Slic3r/bin/../dll/libperl.so(perl_run+0x203)[0x7f4808652ba3]
slic3r[0x403064]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f48074e5830]
slic3r[0x401b69]
```
[slic3r-crash.zip](https://github.com/alexrj/Slic3r/files/247591/slic3r-crash.zip)

Dema, 2016-05-13:
And I have the similar exception in the almost the same spot.
tarting program: /usr/bin/perl /usr/bin/slic3r.pl --gui smartrapcore.stl
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffdbea9700 (LWP 10430)]

Thread 1 "perl" received signal SIGSEGV, Segmentation fault.
0x00007ffff571ab6c in stl_add_facet (stl=0x4589400, new_facet=0x7fffffffc670) at src/admesh/connect.c:960
960       stl->facet_start[stl->stats.number_of_facets] = *new_facet;
(gdb) bt full
#0  0x00007ffff571ab6c in stl_add_facet (stl=0x4589400, new_facet=0x7fffffffc670) at src/admesh/connect.c:960
        new_facet = 0x7fffffffc670
        stl = 0x4589400
#1  0x00007ffff571b127 in stl_fill_holes (stl=stl@entry=0x4589400) at src/admesh/connect.c:918
        facet = {normal = {x = 0, y = 0, z = -1}, vertex = {{x = -51.6128273, y = -88.1715698, z = 1.13686838e-13}, {x = -51.5328445, y = -88.3212128, z = 1.13686838e-13}, {
              x = -51.3943634, y = -88.7777176, z = 1.13686838e-13}}, extra = "\000\224"}
        new_facet = {normal = {x = 4.93677449e-41, y = 4.92542397e-41, z = 4.93649423e-41}, vertex = {{x = -51.6128273, y = -88.1715698, z = 1.13686838e-13}, {x = -51.5328445, 
              y = -88.3212128, z = 1.13686838e-13}, {x = -52.1797142, y = -87.1110077, z = 1.13686838e-13}}, extra = "\000\224"}
        neighbors_initial = {-1, -1, 62}
        edge = {key = {3260069895, 3266197718, 704643072, 3259990992, 3266271526, 704643072}, facet_number = 35259, which_edge = 2, next = 0xc1a00000}
        first_facet = 61
        direction = <optimized out>
        facet_num = <optimized out>
        vnot = <optimized out>
        next_edge = <optimized out>
        pivot_vertex = <optimized out>
        next_facet = <optimized out>
        i = 61
        j = <optimized out>
        k = <optimized out>
#2  0x00007ffff56dd6b5 in Slic3r::TriangleMesh::repair (this=0x4589400) at src/libslic3r/TriangleMesh.cpp:128
[crash2.zip](https://github.com/alexrj/Slic3r/files/263347/crash2.zip)

vlast3k, 2016-05-13:
With 1.3.0-dev version it slices, and produces the attached file.
But in fact slic3r complains that the object is not manifold and there are
many error
it does not even display well

Regards,Vladimir

lordofhyphens, 2016-05-17:
Fix the model. If you can replicate the crash with a manifold model (run it through netfabb online), then I'll remove the "not a bug" label.

Dema, 2016-05-18:
Well, it was a bug in 1.2.9. Non-manifold model doesn't mean that slicer may crash with SIGSEGV :-)

sebastien, 2016-05-18:
For reference, Cura has no problem with it, as is.

lordofhyphens, 2016-05-18:
@Dema Non-manifold model means that all bets are off :-)