slic3r/Slic3r #3318: Slic3r crashes on startup when not using --no-platter, Linux x86_64

RedChops, 2016-05-05:
### Version
1.2.9-243-g510ca9f

### Behavior
After pulling down a fresh git clone, running "perl Build.Pl" and "perl Build.Pl --gui", Slic3r will immediately segfault on execution. I can run it just fine with --no-platter. This is under Gentoo, but I allow the Build script to pull in the libraries required. Stack trace from the segfault is below. I'm not aware of any other perl programs having issues, but to be honest I think this is the only perl based application I use.

Program received signal SIGSEGV, Segmentation fault.
0x00007fffe540612a in wxGLCanvas::Create(wxWindow*, wxGLContext const*, wxGLCanvas const*, int, wxPoint const&, wxSize const&, long, wxString const&, int*, wxPalette const&) () from /usr/lib64/libwx_gtk2u_gl-2.8.so.0
(gdb) bt
#0  0x00007fffe540612a in wxGLCanvas::Create(wxWindow*, wxGLContext const*, wxGLCanvas const*, int, wxPoint const&, wxSize const&, long, wxString const&, int*, wxPalette const&) () from /usr/lib64/libwx_gtk2u_gl-2.8.so.0
#1  0x00007fffe5406329 in wxGLCanvas::wxGLCanvas(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&, int*, wxPalette const&) ()
   from /usr/lib64/libwx_gtk2u_gl-2.8.so.0
#2  0x00007fffe560e304 in XS_Wx__GLCanvas_newDefault(interpreter*, cv*) () from /usr/local/lib64/perl5/5.20.2/x86_64-linux-thread-multi/auto/Wx/GLCanvas/GLCanvas.so
#3  0x00007ffff7aea0eb in Perl_pp_entersub () from /usr/lib64/libperl.so.5.20
#4  0x00007ffff7ae2976 in Perl_runops_standard () from /usr/lib64/libperl.so.5.20
#5  0x00007ffff7a6be01 in Perl_call_sv () from /usr/lib64/libperl.so.5.20
#6  0x00007fffe560d2d0 in XS_Wx__GLCanvas_new(interpreter*, cv*) () from /usr/local/lib64/perl5/5.20.2/x86_64-linux-thread-multi/auto/Wx/GLCanvas/GLCanvas.so
#7  0x00007ffff7aea0eb in Perl_pp_entersub () from /usr/lib64/libperl.so.5.20
#8  0x00007ffff7ae2976 in Perl_runops_standard () from /usr/lib64/libperl.so.5.20
#9  0x00007ffff7a6c075 in Perl_call_sv () from /usr/lib64/libperl.so.5.20
#10 0x00007ffff36ad4a5 in XS_Wx___App_Start(interpreter*, cv*) () from /usr/local/lib64/perl5/5.20.2/x86_64-linux-thread-multi/auto/Wx/Wx.so
#11 0x00007ffff7aea0eb in Perl_pp_entersub () from /usr/lib64/libperl.so.5.20
#12 0x00007ffff7ae2976 in Perl_runops_standard () from /usr/lib64/libperl.so.5.20
#13 0x00007ffff7a7369e in perl_run () from /usr/lib64/libperl.so.5.20
#14 0x0000000000400e39 in main ()

lordofhyphens, 2016-05-05:
From the trace looks like it's somewhere in the interaction between Wx and
OpenGL (the plater uses OpenGL). I don't see this on my Debian builds; not
entirely sure what I can do to help/debug.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

RedChops, 2016-05-05:
Interestingly enough, if I use the precompiled binaries it works okay...I guess this should be closed then as an issue with my build system. Still strange though

closed 2016-05-17 by lordofhyphens