slic3r/Slic3r #3319: Volumetric E support question.

pbrgle1, 2016-05-06:
Hi. I need some explanation of how volumetric e works and how to set the diameter.
Is it enough setting the diameter in the filament tab or do i also need M200 Dxxx Tx in start gcode?
Which one? Or both?
I use Marlin RC5 and the latest slic3r on windows 10.

lordofhyphens, 2016-05-06:
Marlin only understands gcode, so M200 D[filament_diameter_0] (use whatever
shows up in the saved filament profile as the variable name) in your start
gcode is required if you want to set that at the start of every print.

I *think* M200 gets saved to EEPROM on Marlin though, so you could also
hand-set it if your filament diameter isn't changing from roll to roll.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-06-20:
Going to close this.

closed 2016-06-20 by lordofhyphens