slic3r/Slic3r #3320: Spiral: Bad Infill Region on Bottom Layers

CCS86, 2016-05-08:
### Version
1.2.9

### Operating system type + version
Win10 / Repetier Host

### Behavior
Infill region in bottom layers does not properly update when outer perimeter changes. This results in a complete lack of infill:perimeter connection.

[image]
[image]
[image]
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[DoggyVase2.zip](https://github.com/alexrj/Slic3r/files/254094/DoggyVase2.zip)

CCS86, 2016-05-08:
[DoggyVase2.gcode.txt](https://github.com/alexrj/Slic3r/files/254098/DoggyVase2.gcode.txt)

lordofhyphens, 2016-05-09:
Need a screenshot of this from Slic3r itself, not RepetierHost.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

CCS86, 2016-05-09:
I printed that code. It did exactly as you see there. I uploaded the model in the first post, if you want to check it out.

bubnikv, 2016-05-09:
This is an intended behavior.

If you had the infill lines connected along the perimeters where the
perimeters are far from being perpendicular to the infill lines, you would
overfill heavily. The threshold of when to connect the infill lines and
when not is really a matter of taste or an optimization criterion of
overfill against a good connection to the perimeters.

Vojtech

CCS86, 2016-05-09:
@bubnikv, that is not at all what is going on here.

These pictures are of the same region, subsequent layers. Then angle between the infill and perimeter is not changing, yet the gap between them grows in each layer.

xoan, 2016-05-10:
This couldn't be an intended behaviour.

It occurs in Spiral vase mode when the next layers are bigger than the previuos ones. Typically, you can generate an inverted truncate cone (attached) and slice it with the -default- configuration, but activating only the Spiral vase option and by increasing the number of bottom solid layers to see the bug.

[image]

[cone.tar.gz](https://github.com/alexrj/Slic3r/files/256815/cone.tar.gz)

lordofhyphens, 2016-05-17:
@CCS86  You never uploaded configuration you saw the issue with. Please do so.

Also, the model is being auto-repaired on my copy of 1.2.9. which means there's probably something wrong with it. Run it through netfabb at least.

CCS86, 2016-05-17:
I'll export it when I get home.

Does that mean you aren't seeing this behavior when slicing this model?

CCS86, 2016-05-17:
Meshlab says it looks fine:

[image]

CCS86, 2016-05-18:
[config.ini.txt](https://github.com/alexrj/Slic3r/files/269436/config.ini.txt)

CCS86, 2016-05-26:
?

lordofhyphens, 2016-05-27:
Well, 1.3.0-dev still says it had to repair 1000 "things" with the file. Passed it through the online netfabb repair and slic3r is happy.

Messing around with the model and the slicing settings, what it looks like is that it's expecting all of the bottom layers to have the same width all the way up; that's why you aren't seeing them. If you put the infill up to 20% that gets filled in (which of course requires that you turn off spiral vase).

I'm not entirely sure 6 layers on the bottom is actually a good idea (you're not getting the benefit of spiral vase down there). The parser could also be derping on the bottom slope on that model. I don't use deltas or spiral vase though, so I'm not the best source for "what is a good idea" on a delta.

lordofhyphens, 2016-05-27:
@alexrj what looks like is happening is that the "bottom solid" infill goes down, then the space between that and the perimeter is looked at to get filled by regular infill (trying it without spiral vase). Those solid infill items are basically stacked on top of each other and occupy the same space.

CCS86, 2016-06-24:
Any plans for a fix?

lordofhyphens, 2016-06-24:
It's on the list; my own priority is to fix things that are easy to fix
first.

dwillmore, 2019-01-15:
I'm going to leave my comments here as this seems to be the oldest issue.  There is also #4225 and #4396 which are likely the same problem.

What I found is that there is logic in Slic3r to detect the top and bottom of objects.  It determines this by looking at successive layers and seeing if there is anything below it.  If not, that area is classified as 'bottom'.  It looks to see if there is anything above.  If not, then that area is classified as 'top'.  Then there is a process that goes through all of the layers and says "is there 'bottom' in the layer below us?  If so, make the same area of this layer 'bottom'."  It does the same for 'top'.  These areas propagate from layer to layer *throughout the whole region*! 

Here is an example of the bug:
[image]

Note that the colums formed by the top and bottom in the figure to the rigth overlap while there is a gap between the colums in the figure on the left.  Let's look at the layer view for a few mm up from the bottom:
[image]

You can see that the object on the rigth has bottom solid all the way to the perimeter.  But that's not bottom solid!  That's top solid because it's in the colum formed by the top layer.  The object to the left has a gap between the bottom solid and the perimeter showing the bug.