slic3r/Slic3r #3321: internal horizontal wall not supported by any support

CarlosGodoy, 2016-05-09:
### Version 1.2.9
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
not getting support got the walls of an internal horizontal hole in the body
I defined the infill as honeycomb with just 3%, because it would be better than no infill. 
the problem is that the walls of that hole is not resting in an infill wall, and I think support is created only externally of the part, it it got none support.
as it can be noticed in the loaded photo, the base for the hole that got aerial is not resting on an internal infill wall because it is in the middle of a honeycomb.
so it shout build a support underneath, but although is configured to use support, I think support will happen only externally of the part
 
 

[image]

[image]

pyrophreek, 2016-05-10:
What you probably want to try is only infill where necessary, and increase the infill %.  Also, enable extra perimeters if needed.

Unless you have those all enabled, the slicer is behaving properly.

K

CarlosGodoy, 2016-05-24:
dont you think it should make a better job by allowing me to use a small fill pecentage but still supporting something that is necessary? See that because it was selected a small filling percentage, the honey compb is large, wich in that case is the wanted choice. 
See that I will be obligated to make a thicker infill , when I wanted just 5% and still have suport internal support for what is needed

lordofhyphens, 2016-05-26:
@CarlosGodoy, please add the information to the issue as both the
contribution guidelines said and the template for issue reports said. Can't
do anything without a reproducible failure case.