slic3r/Slic3r #3322: Solid infill is not quite solid

VanessaE, 2016-05-10:
### Version
git commit 510ca9f9e2f0e550ee3733510f7738e6ca0128aa with LoH's debian patch

### Operating system type + version
Debian 8.2

### Behavior
The title actually says it all - solid infill is not being printed as solid anymore.  It's still the usual rectilinear pattern, but there are visible gaps between the lines.  Seems to be only about 60 or 70 percent as dense as it ought to be.

#### STL/Config (.ZIP) where problem occurs
[fan-shroud-squirrel-cage-dual-5015-parts.zip](https://github.com/alexrj/Slic3r/files/256704/fan-shroud-squirrel-cage-dual-5015-parts.zip)
[fan-shroud-squirrel-cage-dual-5015-parts.gcode.zip](https://github.com/alexrj/Slic3r/files/256705/fan-shroud-squirrel-cage-dual-5015-parts.gcode.zip)
[Prusa i3 - Slic3r 1.3.0-dev config bundle.ini.zip](https://github.com/alexrj/Slic3r/files/256708/Prusa.i3.-.Slic3r.1.3.0-dev.config.bundle.ini.zip)

Mikk36, 2016-05-10:
This looks like a machine configuration issue.
Try printing a lot slower and see if it helps.
Also set the default extrusion width to the same as nozzle diameter. This will help you diagnose your issues better.

VanessaE, 2016-05-10:
I might have suspected that, if not for the fact that this machine's quite well-calibrated and has been for a while now.

In any case, I recall reading somewhere that one should set the print width at 25% over the nozzle diameter, or larger, so 0.5mm for my 0.4mm nozzle.  Setting all of the "advanced" size settings to that value does not help.

xoan, 2016-05-10:
If the preview tab shows the desired result and later, it isn't the same in the printed part, it is a hardware problem.

We all need to understand that Slic3r is not intended as a workaround for all of the machine related problems, so this problems should be fixed first at the hardware level.

VanessaE, 2016-05-10:
Actually, looks like you guys were right, it was a hardware failure.  *sigh*

closed 2016-05-10 by VanessaE

Mikk36, 2016-05-11:
That 25% over the nozzle diameter is a bit of a myth to be honest. Cura (and Ultimaker in that sense) have been doing just fine with all extrusions always being the same width as the nozzle diameter.

lordofhyphens, 2016-05-12:
@Mikk36 Slic3r defaults external perimeters to the nozzle width in a lot of cases. 

The reasoning is that wider traces give better fusion for the most part (2.2 seems to be the "magic" ratio that was bandied about).

Mikk36, 2016-05-12:
Only if you print with high enough heat. If you're just about getting good bonding with the outer perimeters, then you might get a lot worse bonding with the fatter traces.