slic3r/Slic3r #3324: boost build error

samiaruda, 2016-05-11:
### Version
git repository

trying to run slic3r from github

### Operating system type + version
Windows 10 64-bit

### Behavior
have downloaded boost libraries from boost.org (v 1.60.0) and placed the unzipped content into C:\dev\boost_1_60_0

when i go to command prompt and type in bootstrap.bat mingw i receive the following error
-----
Building Boost.Build engine
'gcc' is not recognized as an internal or external command,
operable program or batch file.

Failed to build Boost.Build engine.
Please consult bootstrap.log for further diagnostics.

You can try to obtain a prebuilt binary from

   http://sf.net/project/showfiles.php?group_id=7586&package_id=72941

Also, you can file an issue at http://svn.boost.org
Please attach bootstrap.log in that case.
-----

later on mingw installation have been done and updated all C++/C related libraries but still receiving same issue

bubnikv, 2016-05-14:
First, please use the boost 1.59. I was unable to compile Slic3r with 1.60 on Windows. Maybe the fix to get Slic3r working with Boost 1.60 on Windows is trivial, but I did not have time to investigate it further.

Second, you have to have mingw installed before you try to compile Boost. The easiest way to get mingw installed is to follow the guidance of Alessandro and install the Strawberry Perl as the Strawberry Perl comes with its own mingw. After the Strawberry perl is installed, you may open cmd and run gcc. It shall work. If not, then you are missing an entry in the PATH environment variable pointing to the strawberry mingw installation folder.

Vojtech

samiaruda, 2016-05-16:
Hi, thank you very much. I have installed strawberry perl and then boost 1.59 worked with no problems.
then I have continued with all commands as described in documentation and completed cloning the project.

I was about to download dependencies and did a check if perl is recognized by typing build.pl command in cmd but received following error;

C:\>Build.PL
'Build.PL' is not recognized as an internal or external command,
operable program or batch file.

Could you have suggestions on this?

Looking forward to it,
Best, arda

vlast3k, 2016-05-17:
try
perl build.pl
if it fails then perhaps the path to perl is not set correctly

samiaruda, 2016-05-17:
hi, perl -v gives reasonable verification. when typed perl build.pl or build.pl following messages are faced. your help is appreciated,

C:\Program Files\Git\Slic3r>perl -v
This is perl 5, version 22, subversion 2 (v5.22.2) built for MSWin32-x64-multi-thread
Copyright 1987-2015, Larry Wall
Perl may be copied only under the terms of either the Artistic License or the
GNU General Public License, which may be found in the Perl 5 source kit.
Complete documentation for Perl, including FAQ lists, should be found on
this system using "man perl" or "perldoc perl".  If you have access to the
Internet, point your browser at http://www.perl.org/, the Perl Home Page.
-----
C:\Program Files\Git\Slic3r>perl build.pl
Can't spawn "cmd.exe": No such file or directory at build.pl line 113.
Can't spawn "cmd.exe": No such file or directory at build.pl line 117.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Don't worry, this module is optional.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Don't worry, this module is optional.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Don't worry, this module is optional.
Can't spawn "cmd.exe": No such file or directory at build.pl line 139.
Can't spawn "cmd.exe": No such file or directory at build.pl line 159.
The XS/C++ code failed to compile, aborting

lordofhyphens, 2016-05-17:
@samiaruda Everything I've seen so far is telling me you've got some strange environment/PATH issues.

samiaruda, 2016-05-23:
I just switched to win7 32-bit machine and everything worked smoothly.
I guess we can close this.
thank you very much for your guidance.

closed 2016-05-23 by samiaruda