slic3r/Slic3r #3325: No slowdown

MacFlieger, 2016-05-12:
### Version
1.2.9-243-g510ca9f

### Operating system type + version
OS X 10.11.4

### Behavior
I have enabled "Auto cooling" and slow down printing if the print time is below 20s with minimum print speed of 1mm/s.
With this configuration the print slows down to approximately 20s on small objects if I am using Slic3r 1.2.9.

But if I am using the above mentioned actual developer version there is no slowdown.

It is easy to replicate, even as dry run:
- cube with 20mm size
- perimeter:1, no infill
- speed for perimeter 60mm/s
- one layer is 80mm long

- without auto cooling:
layer time around 2s

with auto-cooling (slow down if below 20s, min print speed 1mm/s):
layer time is again around 2s (no slow down), but should be around 20s.

Mikk36, 2016-05-12:
This was supposedly fixed with 510ca9f9e2f0e550ee3733510f7738e6ca0128aa

MacFlieger, 2016-05-12:
I know, but I am using this version [1.2.9-243-g510ca9f](https://github.com/alexrj/Slic3r/commit/510ca9f9e2f0e550ee3733510f7738e6ca0128aa).

Mikk36, 2016-05-12:
So what's your point, if it has been fixed already and will be incorporated into the next release?

MacFlieger, 2016-05-12:
As stated I am using this fixed version [510ca9f](https://github.com/alexrj/Slic3r/commit/510ca9f9e2f0e550ee3733510f7738e6ca0128aa), but there is no slowdown.
I have expected, that if I am using this fixed version, the slowdown is working.

Mikk36, 2016-05-12:
Oh blimey, I must have been blind not noticing the checksum version being the same.
Sorry for the confusion :)

Mikk36, 2016-05-12:
Can't replicate the issue though.
https://dl.dropboxusercontent.com/u/54281287/20mm-box%20%282%29.gcode

Also, how are you seeing your version as being 1.2.9?
It should be 1.3.0 already.

MacFlieger, 2016-05-12:
Strange.
This is my GCode for a 20mm Cube, 1 Perimeter, no infill, no top/bottom solid, 60mm/s, auto cooling 20s down to 1mm/s.
[slowdown.gcode.zip](https://github.com/alexrj/Slic3r/files/261031/slowdown.gcode.zip)

The speed for the perimeters is always 60mm/s (F3600).

> Also, how are you seeing your version as being 1.2.9?
> It should be 1.3.0 already.

The GUI of Slic3r and the GCode say 1.3.0-dev
The "1.2.9-243-g510ca9f" is the output of the command "git rev-parse HEAD".

Mikk36, 2016-05-12:
Are you sure you've recompiled Slic3r after updating the files from git?

MacFlieger, 2016-05-12:
Yes I have recompiled several times. I have checked, that the changes in "CoolingBuffer.pm" are present.
And compiled with "perl Build.PL" and "perl Build.PL --gui"

Sorry with the number, my error:

"git rev-parse HEAD":
510ca9f9e2f0e550ee3733510f7738e6ca0128aa

"git describe --tag"
1.2.9-243-g510ca9f

Mikk36, 2016-05-12:
Ok, seems as if the bug happens if you disable verbose output.
Enable verbose output and the slowdown kicks in.

MacFlieger, 2016-05-12:
Yes! :)
If I enable verbode output, than the slow down works.
Thanks for this workaround.

But I think, this is really a bug, or?

Mikk36, 2016-05-12:
Yes, that is a bug. The slowdown should kick in regardless of verbose output option.

lordofhyphens, 2016-05-12:
@MacFlieger Try the branch that @hyperair just referenced this issue with.
https://github.com/hyperair/Slic3r/tree/fix-cooldown-verbose

MacFlieger, 2016-05-12:
Yes, the branch of hyperair is working.

PR #3326 (2016-05-19): Fix cooling not working if !gcode_comments
The cooling markers were being passed into GCodeWriter::set_speed() as a
comment which were being ignored if gcode_comments was false.

Fixes: #3325
commit 209b919f (2016-05-19): Fix cooling not working if !gcode_comments

The cooling markers were being passed into GCodeWriter::set_speed() as a
comment which were being ignored if gcode_comments was false.

Fixes: #3325
closed 2016-05-19 by lordofhyphens