slic3r/Slic3r #3327: Feature Request: ungroupped STL export to save one STL per Object

realthor, 2016-05-13:
### Version
latest/ALL

### Operating system type + version
Windows, Linux

_Is this a new feature request?_
YES

Hello,
I and many of my designer friends would love the ability to break a STL in several files each containing the Parts that the program identified when the Ungroup was issued.
Please have a checkbox on export to export each Object to a file.

bubnikv, 2016-05-14:
If you select an object, click the right mouse button, a popup menu will appear.
The last menu selection is what you need: "Export object as STL".

Vojtech

realthor, 2016-05-14:
I have about 50 objects in my design :(... and I export them once a week to
github ... saving each would waste quite some time... What about "select
all" then "save each to STL"?

bubnikv, 2016-05-14:
There is also a tool available from Alessandro to do exactly what you need: to split one stl into multiple files.

I believe the tool is being installed on Ubuntu Linux with the Slic3r package. On other systems, you need to download the Slic3r sources, compile Slic3r and run

perl Slic3r\utils\split_stl.pl

Vojtech

realthor, 2016-05-14:
Thanks so much, I'll have a look at it.

alexrj, 2016-05-14:
Thanks @bubnikv

alexrj, 2016-05-14:
Actually, @bubnikv and @realthor, you can do that thing using the command line interface of Slic3r itself using the `--split` argument:

```
./slic3r --split file.stl
```

I realize that the `--split` argument is not documented in http://manual.slic3r.org/advanced/command-line so I'll reopen this one as a reminder. Having a command for saving all parts as individual parts, as the original request says, actually makes sense.

realthor, 2016-05-19:
Please also add this in the contextual menu when one selects more than an object and right clicks ...