slic3r/Slic3r #3329: Can't run 1.2.9 on Linux Mint 17.1, nor can I build from source.

JoshMcCullough, 2016-05-17:
### Version
1.2.9 x64, also #510ca9f9e2f0e550ee3733510f7738e6ca0128aa

### Operating system type + version
Linux Mint 17.1 

### Behavior
Unable to run this version from the pre-compiled package, I get this error:

```
josh@Labby ~/Slic3r $ bin/slic3r  --gui
Can't load '/media/data1/home/josh/Slic3r/lib/std/auto/Wx/Wx.so' for module Wx: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory at <content>/DynaLoader.pm line 157.
 at /media/data1/home/josh/Slic3r/lib/std/Slic3r/GUI/AboutDialog.pm line 6
Compilation failed in require at /media/data1/home/josh/Slic3r/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /media/data1/home/josh/Slic3r/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /media/data1/home/josh/Slic3r/lib/std/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /media/data1/home/josh/Slic3r/lib/std/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 71) line 1.
```

Building from source (perl Build.PL --sudo), I get:

```
--> Working on ./xs
Entering /media/data1/home/josh/Slic3r-source/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (1.59)
Checking if you have Module::Build 0.38 ... Yes (0.4003)
Running Build.PL
Configuring /media/data1/home/josh/Slic3r-source/xs ... Can't locate ExtUtils/CppGuess.pm in @INC (you may need to install the ExtUtils::CppGuess module) (@INC contains: /etc/perl /usr/local/lib/perl/5.18.2 /usr/local/share/perl/5.18.2 /usr/lib/perl5 /usr/share/perl5 /usr/lib/perl/5.18 /usr/share/perl/5.18 /usr/local/lib/site_perl .) at Build.PL line 7.
BEGIN failed--compilation aborted at Build.PL line 7.
N/A
! Configure failed for ./xs. See /home/josh/.cpanm/work/1463444921.21632/build.log for details.
The XS/C++ code failed to compile, aborting
```

lordofhyphens, 2016-05-17:
I know about the CppGuess problems, it's also on the Travis build server. I
think that something happened to CppGuess or something, but I don't really
know at this point.

The prebuilt binary wants the GTK-2.0 X11 lib, which is apparently not on
your system.
Get my Debian branch and build your own package for it; that should work a
little better.

Make sure you have everything needed to build packages (if slic3r is in
mint's repo, apt-get build-dep slic3r will help), then run "fakeroot
debian/rules binary" from my branch.

Or you can try my debian package:
https://drive.google.com/open?id=0B3aPl7CmzLRvUkpiYWxLYnEyVmM
It's built for Debian Jessie though.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

JoshMcCullough, 2016-05-17:
Thanks. I saw this post but not sure really how to add this arg during link: http://stackoverflow.com/questions/11939934/c-apache-module-fails-on-znss4-rep20-s-empty-rep-storagee

I'll try your branch and see how it goes.

lordofhyphens, 2016-05-19:
Can this be closed @JoshMcCullough ?

JoshMcCullough, 2016-05-19:
I haven't tried your branch yet, but regardless, it's broken still -- until some fix is merged/pulled in which makes it so I can build directly on Ubuntu-based OS'. (I'd leave it open.)

lordofhyphens, 2016-05-19:
95% sure it isn't slic3r itself, and using build.pl on Ubuntu can break
your perl install and do other weird things

lordofhyphens, 2016-07-04:
@JoshMcCullough on the command line, 
`export LDLOADLIBS=-lstdc++` before you run perl Build.PL to avoid a bug in the current build of MakeMaker (upstream problem :( )

alexrj, 2017-03-17:
I think we can close this one as it's pretty old. We've done a lot of changes and improvements. @JoshMcCullough, in case you need more help feel free to reopen.

closed 2017-03-17 by alexrj