slic3r/Slic3r #3331: Many Hole after slice

Cerya, 2016-05-17:
### Version
Slic3r 1.2.9

### Operating system type + version
Windows 7  ultimate 32bit

### Behavior
i have many hole, when i'm trying to slice this STL file with slic3r.
We have tested to repair with netfabb and result is same.

on top we can see many hole:
[image]

Original STL file
[image]

simple test with Cura
[image]

#### STL/Config (.ZIP) where problem occurs
[90mm (repaired).zip](https://github.com/alexrj/Slic3r/files/267966/90mm.repaired.zip)

lordofhyphens, 2016-05-17:
Please provide a screenshot showing Slic3r's preview, not RepetierHost. Also please provide the exported config "File -> Export Config", as you were prompted when writing this issue. 

However, my best guess as to what is going on is that the toolpath just isn't well-resolvable with your define nozzle diameter. There's some really thin walls there and Slic3r tries to form loops with the toolpath. This is the downloaded model with a 0.35mm nozzle defined (which is one of the ones I have). 0.4 also seems to work. I suspect it's a combination of bug (weird edge case) and your model's design.

[image]

See, the "hole" is the start and end of the loop and then the single wall. There probably isn't enough room to fill those gaps in and then the logic is skipping over them a bit.
[image]

lordofhyphens, 2016-05-17:
I've not checked with this with the current -dev version, which has the variable width perimeters feature in it. Screenshots above are from the binary release of 1.2.9.

Cerya, 2016-05-18:
Yes, you have found problem.

It's STL file because when i follow your instruction, slice is good.

I have downloaded this files so i think is just a problem on design no on slic3r.

Thanks for the tips for nozzle diameter.

closed 2016-05-18 by Cerya