slic3r/Slic3r #3332: Solid Fill of a layer depends on upper layer (bug + workaround)

vlast3k, 2016-05-18:
### Version
1.3.0_dev (current)

### Operating system type + version
Windows 7 64-bit

### Behavior
In some strange way the rendering of a solid layer depends on what is on top of this layer, regardless of the number of top and bottom shells.
This took me quite some time to figure out, and the great thing is that there is a workaround, that could eventually help others.

I am trying to slice an object of few parts, and one of them looks like this
[image]
the bottom part is 0.2mm high, the wider part is 0.2 mm and the top squares are also 0.2 mm. (and layer height is 0.2 mm)

Now if i import this in Slic3r and slice it initially, i get this for the layer with the wider part
[image]

notice that the solid fill is in squares according to the places where on the top there are the squares.

now if i go in layer configuration and set the Horizontal Shells to something different (e.g.4) and then set them back to the original value (3), the solid infill is invalidated and now when i reslice i get what is expected:

[image]

Notice that now the solid fill is consistent :)

This works perfectly fine for me (now that i know the workaround). If it can be fixed - it would be nice, but if not - i hope this will save some time to some other fellow slicer :)

#### STL/Config (.ZIP) where problem occurs
this is hte export as AMF (3 objects) and the config, in case anyone would like to have a look
https://www.dropbox.com/s/rhbgbo2hrtfkayf/example.ZIP?dl=0

lordofhyphens, 2016-05-18:
Neat bug there. 
FYI, you can upload zip files onto github issues via drag-and-drop.

lordofhyphens, 2016-05-19:
Could you get the git commit revision for the version of 1.3.0-development you are working on? Thanks, it is important for reproduction purposes.

vlast3k, 2016-05-19:
I synced it 1 week ago, but strangely when i type the git-describe tag it tells 1.2.9, but in the gui, the version is 1.3.0_dev... 
btw.. i had tried to include the ZIP but somehow github still complained so i put it on dropbox

below is the output

```
C:\develop\Slic3r>git status
On branch master
Your branch is up-to-date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be

        xs/compilet.base
        xs/compilet.def
        xs/compilet.exp
        xs/compilet.lds

nothing added to commit but untracked files present (

C:\develop\Slic3r>git describe --tag
1.2.9-243-g510ca9f

C:\develop\Slic3r>git rev-parse HEAD
510ca9f9e2f0e550ee3733510f7738e6ca0128aa

C:\develop\Slic3r>
```

lordofhyphens, 2016-05-19:
Hrm, github may be freaking out over the capital ZIP extension instead of
lower case? Now I am curious.

lordofhyphens, 2016-05-19:
https://github.com/alexrj/Slic3r/commit/510ca9f9e2f0e550ee3733510f7738e6ca0128aa

Was the commit. I believe that the 1.3.0 tag hasn't been made yet