slic3r/Slic3r #3333: Graduated Infill

JoshMcCullough, 2016-05-19:
This is a general suggestion for a new feature called "graduated infill". This feature would accomplish three goals:

1. Reduce print time.
2. Reduce material use.
3. Attain better top-layer adhesion/quality.

## Theory

We can achieve these three goals by setting a low infill and enabling a new option called "graduated infill". Which will gradually increase the fill density, so as to provide more support for the first and subsequent top layers of fill.

## Example

Assume we are printing a solid cube which is 16 layers tall and has 3 layers of solid fill on the top and bottom (10 layers of infill). We have set the infill density to 12.5%.

### Current Outcome

_Layers 1-3_ are printed with 100% fill.
_Layers 4-13_ are printed with 12.5% fill.
_Layers 14-16_ are printed with 100% fill.

**Result:** The top of the cube is likely riddled with holes, not smooth at all, and is very brittle.

### Suggested Outcome

_Layers 1-3_ are printed with 100% fill.
_Layers 4-10_ are printed with 12.5% infill.
_Layer 11_ is printed with 75% infill.
_Layer 12_ is printed with 87.5% infill.
_Layers 13-16_ are printed with 100% fill.

**Result:** The top of the cube is solid, smooth and strong.

[image]

### Comparing Results

To obtain the "suggested" results with the current functionality, the infill would have to be at least 30-40% (probably higher).

## Practice
When enabled, Slic3r will determine the best way to "close the infill gap" when nearing the top of an object or portion of an object. Slic3r will gradually increase the fill density, so as to provide more support for the first and subsequent top layers of fill.

### Options

In addition to the new option to enable this functionality, it may be wise to allow the user to adjust the graduation rate. In the example above, the graduation rate is .125 (12.5%) which appears to provide a good bond with the next/prior layer of infill.

lordofhyphens, 2016-05-19:
So how does this work with infill pattErnst that aren't rectilinear? 

All infill is currently generated through Perl, so it should be feasible for you to hack something in (as a permanent modification) to show how it would behave in real world terms. If it looks promising I can do the rest of the plumbing to make it a tweakable option.

bubnikv, 2016-05-19:
Kisslicer already does something like this. It puts a 50% rectilinear infill in the layer directly below the 1st closing layer. This behavior is highly desirable.

JoshMcCullough, 2016-05-19:
@lordofhyphens It would work the same regarding other patters. E.g. Honeycomb would gradually create smaller honeycombs as it nears the top of that part of the object. Other patters could do the same, but I'm not sure how the more advanced patterns would "line up" when the infill density is changed. I could draw up a diagram of what this might look like for honeycomb. It would require Slic3r to ensure the patterns are fixed, starting at the center of the object (or something) to ensure that higher density fills lie atop the lower density fills correctly.

I may be able to take a crack at it but success it not that likely. :)

mrvn, 2016-05-24:
You get the same or probably better effect by simply selecting 5 top fill layers.

I think the bigger problem is that when slic3r changes from infill to top fill for only part of the area then it does not extend the top fill region to rest on infill. With small infill percentage the sides of the top fill then simply fall into the holes between the infill.

JoshMcCullough, 2016-05-24:
@mrvn Maybe, but this method would allow you to go much lighter on the "core infill" and only increase the fill as you reach a solid layer. It could by my printer that's just not good at top infill (hand-built Prusa from many years ago -- Robo R1+ on order).

lordofhyphens, 2016-05-29:
Also, similar ideas to this have been proposed several times (which led to the bridging infill solid layer).

Tinchus2009, 2016-06-07:
Right now, Im haveing this behaviour useing modifers. Of course this mean that I have to take the work of building a modifier. What I do is build a modifier to print a low value infill, like 5% , and set the objetct to have 30% for example. What I got in this way, for example with a cube, is to have: bottom layers (3 in my case is enough) then and infill of 5% (with 3 perimeters is strong enough). All this 5% infill area is produced useing a modifier. The object itself is set to have 30% rectilinear infill. So the modifier only cover till layer 90 (as example, this object lets say it has 100 layers). So stating at layer 91, I start getting a 30% rectilinear infill, adn the last 4 top layers gives me a perfect smooth result) I save a lot of printing time this way.

So my suggestion would be to add and option to set the infill % as we have now and a second value to affect the infill % of, lets say, the 5 previous layers BEFORE the first top layers.

lordofhyphens, 2016-06-07:
@Tinchus2009 it's in the PR or bust milestone, which means that if someone wants to do a PR then we'll look it over and probably merge it in.

I added a small OpenSCAD document to make it easier to generate large flat modifier meshes, and the UI for modifier meshes certainly could be improved. I'll try to get a JSCAD version together that works as well for those people who don't want to have OpenSCAD on hand (although IIRC you can copy/paste the openscad script into the openjscad window and it'll work).

You could also add more top solid layers, which are done via bridging as well.

Tinchus2009, 2016-06-07:
@lordofhyphens , I understand, my comment was regarding the tittle "NEED ADVICE AND IDEAS FROM COMMUNITY ".
Where is that openscad document?

lordofhyphens, 2016-06-08:
The util folder in the main slic3r checkout.2

JoshMcCullough, 2016-06-08:
@Tinchus2009 That's basically my suggestion. But I think it would work better to set the infill as you normally would (e.g. 10%) and just have a checkbox to enable "graduated infill" all that would do is decrease the infill as the print head moves up (away from the bottom solid layer), then fill using the set 10% infill ratio, then start gradually increasing infill as the top solid layers are reached.

@lordofhyphens I may have missed the "bridge infill" option as I was running an older version of Slic3r since the current version doesn't seem to build on Ubuntu. (I know, you reached out to me with support regarding that but I haven't got to try your suggestions yet.)

Tinchus2009, 2016-06-09:
Why do it gradually? Doing that way i think you are still wasteing plastic and precious time. A lot of times useing modifier  i use a 3% infill and i jump to 20%, linear infill, and this works really good. The bridging that happens between the 3% and the first 20% infill layer is made without problems (at least with a well calibrated printer), and this procedure saves me 1, 2 and 3 hs of printing sometimes. Im not a perl coder so i cant say much there but i think it would be easier to just have a tick box to have the option to activate a different % infill 5 layer before a top layer. I say 5 because i found this is the number required to have a good base from a % infill change and then start with the top layer.

JoshMcCullough, 2016-06-09:
@Tinchus2009 Can you point me in the direction showing how to implement this? I'm not sure what you mean by "modifier". Thanks!

bubnikv, 2016-06-18:
@lordofhyphens I agree 130% with @alexrj on his proposal #1815. The idea is to always extend the solid infill to a closest anchor, which is either the sparse infill or a perimeter line. The method of kisslicer to always produce a sparse bridging layer below a solid layer makes sense as well, probably more sense than the proposed gradual infill. In my opinion, the two features (a sparse bridging infill and the anchored infill) shall be automatic, therefore no UI required.

Regarding the gradual infill. One shall keep in mind, that the FFF method produces objects with different properties in the Z axis than in the XY plane. For the infill it is important, that the sparse extrudates are supported one by each other, otherwise they will not produce a strong enough bond. In my opinion, features like the 3D infill are of little practical use, because the infill patterns do not overlap fully, therefore producing a weak bond. Therefore the gradual infill is problematic as it is difficult to design a 3D structure, which would produce sufficient intra-layer bonding without wasting filament and print time.

mrvn, 2016-06-20:
Only way I can think of to make sure of getting a good overlap is to double the infill from one layer to the next. That way lines of the new infill would land perfectly on lines of the old. At least where possible. But I also agree with #1815. It's the lack of anchoring that is the problem. It doesn't need a gradual transition. A sparse bridging layer could save time and filament. Currently I just have more top fill layers than bottom and get good results where that finds anchors.