slic3r/Slic3r #3334: Ooze prevention in printer settings?

smartroad, 2016-05-21:
Currently Ooze prevention is listed in the Print Settings under "Multiple Extruders". I was thinking, would it not be better to place it under print settings? I have a printer with 2 extruders, but I often only use one so have set two print settings, one for the single extrusion and one for dual. Given that Ooze prevention is only needed when using more than one extruder, would it be an idea to have it listed under the "Capabilities" in the Printer Settings where you set for more than one extruder?

Just a thought :) This could just be something specific to my use but I currently need to have 2 sets of every Print Settings to account for the Ooze prevention if I am only printing using a single extruder.

lordofhyphens, 2016-05-26:
@smartroad you mean "print-er" settings, right (for where it goes). 
I suspect it went in here because it made sense at the time. Unless @alexrj or someone else can provide a decent explanation of why it should be in Print Settings I can probably put together a small PR that moves it over. It would cause some issues though with existing profiles so far as I can tell.

lordofhyphens, 2016-05-27:
@smartroad there it is. If you're running slic3r from source you can use my patch to change your local copy in the meantime.

lordofhyphens, 2016-05-29:
Going to close this as there's a PR now to change it.

alexrj, 2017-03-12:
It's in Print Settings because you might want to use it for nicer prints and keep it disabled for draft quality prints or when you use an extruder for soluble support material.
I understand the issue raised by @smartroad and I think we should just make sure the ooze_prevention logic is ignored when the print uses just one extruder. Many things related to multiple extruders are already smart enough to trigger only if the print actually uses more than one.

Patola, 2017-03-12:
Is this why the ooze shield is not being generated? I tried several versions of slic3r and noticed that the 'high skirt' is not generated at all.

Patola, 2017-03-12:
@alexrj As I understand the ooze shield is also quite useful as an 'anti-warp' shield, so it makes sense to be used by non-dual-extruder printers. Or should I open a feature request for a proper 'warp shield' instead?

alexrj, 2017-03-12:
@Patola, Ooze prevention is specifically aimed at printing with multiple extruders. If you just want a tall skirt, just set "Skirt Height" to a high value like 999999.

alexrj, 2017-03-12:
Done. Ooze prevention now can be left enabled as it will be ignored when a print uses a single extruder.

closed 2017-03-12 by alexrj