slic3r/Slic3r #3335: Per-object slicing settings

sdaitzman, 2016-05-25:
Is there any way to slice multiple objects to be printed simultaneously with different settings per object? Thanks!!

CC @Louieadamian

lordofhyphens, 2016-05-25:
Yes, right click on the object in the plater and select settings
There is not a way to assign profiles to individual objects; interface is
designed for overriding a few things.

LouieAdamian, 2016-05-26:
Is there a Command line way to do it?

lordofhyphens, 2016-05-26:
@LouieAdamian I don't know; I don't use the command line myself.

h3rb, 2016-05-30:
No there is no way.  This feature (as well as related features of being able to slice REGIONS of the same model with different settings, not just LAYERS, and also LAYERS) should be supported on command line.  It was first requested 5 years ago as far as my research has shown.

lordofhyphens, 2016-05-30:
Help work out an interface for it and it will be closer to being done.

lordofhyphens, 2016-07-05:
As this was a question and said question was answered, closing. Please open a new issue if you have a feature request; going to lock this topic.

@h3rb modifier meshes combined with simple SCAD or JSCAD setups (which are in the utils directory now) cover most of the "change settings whenever you feel like it, regardless of whether or not it's a good idea" cases. I've been looking into making support respect modifier meshes but that's probably going to be very fiddly.

closed 2016-07-05 by lordofhyphens