slic3r/Slic3r #3336: Run another incremental Stable build off of -stable?

lordofhyphens, 2016-05-26:
I've been getting some chatter in #reprap asking for "another stable build". Thing is  the only thing that's happened in -stable that I've noticed is the patch to paper over the gapfill problems. I think the solution from 1.3.0-dev (which is also not "out" yet) is better, but I've seen it have some weirdness combined with autospeed (although I haven't seen that particular bug lately).

@alexrj any thoughts on this? Could you point me at something so I can figure out how to produce windows builds off of HEAD for the people who have been waiting for stuff to test? I figure anything to help people get useful testing in will help us get to a point where we can mark it as "working/stable" (at least w/r/t milestones reached).

bubnikv, 2016-05-26:
Joseph,

we at Prusa Research have found a way to create builds for 32bit Windows, 64bit MacOS (Maveric and up) and some Debian 64bit. We have an automatic build system running, so we can produce a build at each checkin, we can produce builds for any branch, and each build is clearly indentificable. Currently the Windows 64bit builds are missing, but we are working on it.

Our latest builds for Windows and MacOS are contained in our drivers. These builds are based on the 1.3.0-dev top of Alessandro's github.
http://www.prusa3d.com/drivers/

The builds come from our build system in the following form. These are a bit older, but the structure of the builds is the same.
http://shop.prusa3d.com/forum/software-f13/stable-slic3r-builds-t844.html

We can provide the automatic Slic3r builds to the community.

Vojtech

lordofhyphens, 2016-05-26:
Don't your current builds include a stack of patches that is specific to
your local version? Or am I hallucinating that?

Autobuilding from HEAD of alexrj/Slic3r would be convenient, though, we
could link from slic3r.org to prusa3d.org (as "contributed builds"). One
thing I think that would be important, though: can your build system, as a
pre-hook, append the git commit ID (at least in shortened form) to the
version number in the About screen pre-build so that when people file bug
reports we can know which commit they were working off of?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2016-05-26:
> Don't your current builds include a stack of patches that is specific to
your local version? Or am I hallucinating that?

We have both.

> Autobuilding from HEAD of alexrj/Slic3r would be convenient, though, we could link from slic3r.org to prusa3d.org (as "contributed builds"). One thing I think that would be important, though: can [...]

We have that.

Vojtech

lordofhyphens, 2016-05-27:
@bubnikv Well that sounds great to me, what would go into having a download page of just fresh slic3r (without your patches) builds then for ubuntu/win/osx? (my Debian package is a mix of the package information from -stable and current source).

I would suspect that the files at 
http://www.prusa3d.com/downloads/dev/slic3r/#_ga=1.100518558.1042046276.1442479076
are your branch then?

@alexrj any thoughts?

lordofhyphens, 2016-05-27:
In my testing, removing all of the text and entering it as some other number works. The reason it is behaving weirdly like that is because it's pulling the values as enumerations and selecting most of the text, I think, matches the rest. The "simple" fix is to change PrintConfig.cpp to not add the extra text after the label. That forms what could be considered a UI bug though.

I'll do the commit and submit the PR for discussion.

I can't reproduce the crash on 1.3.0-dev (b06861636623b7b7d7ab2ce32d37163d5d33d7c7) on Linux.

lordofhyphens, 2016-06-25:
Looks like I've got a way to make Win32 builds; assuming it works for people who aren't me anyway.

bubnikv, 2016-06-25:
Regarding the Linux builds, that is a problem due to the "Linux DLL hell", which what I have learned, is much, much worse than the Windows DLL hell. We built something on some version of Ubuntu, but frankly I am not quite sure how compatible it will be. One could either compile everything static, or pack half of Linux with Slic3r. I don't like either solution. The Linux build is not a priority for us though, as it is relaively easy to compile Slic3r on Linux.

The MacOS way is to link most of the dependencies static and to have just the very base MacOS libraries shared. These base MacOS libraries are then stable.

Vojtech

lordofhyphens, 2016-07-05:
Still trying to get appveyor to behave -.-;

alexrj, 2017-03-02:
I'm closing this one - I think it's not worth releasing 1.2.10 out of the stable branch, since the master branch got much much forward and it's actually much more stable than the stable branch. 1.3.0 is the goal.

closed 2017-03-02 by alexrj