slic3r/Slic3r #3338: Instability in Slicer

designermx2, 2016-05-27:
### Version
_Version of Slic3r used goes here_
Slicer 1.2.9 for mac

### Operating system type + version
OSX El Captain 10.11.5 (15F34)

### Behavior
- When I am entering the manual value in the fields where supports configuration in contact z distance, to enter the value if I exit the setup and go back or save, the value returns to 0.2 (detachable). Sometimes i need a different value of 0.1, 0.3

*Reproduce problem:
 - Change value and don't remove the (detachable) name. Change the parameters and go back!
All Slic3r previous do that!

Sometimes i have a crash screen.
* _Expected Results_
 - Remove all value when click on the value area. Allow you to use a different value
_Is this a new feature request?_
 - Yes

[Archive.zip](https://github.com/alexrj/Slic3r/files/285744/Archive.zip)

Patola, 2016-05-27:
I also experience this bug in the latest git version of slic3r. If I change only the "0.2 (detachable)" to "0.3 (detachable)" it does not hold the value. After I get out of the field entry it gets back to "0.2 (detachable)".

A workaround for this is editting "0.2 (detachable)" to just "0.3" - removing the " (detachable)" part. But the GUI should be smart enough to get just the numeric value, maybe a $config->set('support_material_contact_distance', $config->{support_material_contact_distance}+0.0); for automatic casting?

lordofhyphens, 2016-05-27:
In my testing, removing all of the text and entering it as some other number works. The reason it is behaving weirdly like that is because it's pulling the values as enumerations and selecting most of the text, I think, matches the rest. The "simple" fix is to change PrintConfig.cpp to not add the extra text after the label. That forms what could be considered a UI bug though.

I'll do the commit and submit the PR for discussion.

I can't reproduce the crash on 1.3.0-dev (b068616) on Linux.

lordofhyphens, 2016-05-27:
Check my pull request and build off of that branch; see if it helps.

lordofhyphens, 2016-07-11:
I am beginning to believe that it would be better to change the UI element to a plain text box instead of the combo box; the combo box is a bit more clever that I am comfortable with.