slic3r/Slic3r #3340: some SolidInfill of first BottomLayer after InterfaceLayer is forgatton

metalzse, 2016-05-27:
### Version 1.2.9
### Windows 7
### Behavior

Some of the solid bottom infill after the last support interface layer will not be filled.

Layer 61:
a. : support interface layer
b. : build material
[image]

Layer 62:
a. : this bottom solid infill is right executed
b. : this bottom solid infill was not excuted. 
[image]

Layer 63:
a. : at this layer, all infill are normal again.
[image]

lordofhyphens, 2016-05-27:
Hello, please include the information requested in the contributing guidelines (and the text you had to erase to write your report).

metalzse, 2016-05-30:
Sorry for leak of information. Here is the full information (I hope I did not miss something again)

### Behavior
* _Describe the problem_

Some of the solid infill of first bottom layer after support interface layer is missing.
Number of bottom layers ist set to 3, but sometimes, only part of first bottom layer after support interface layer is filled. 
* _Steps needed to reproduce the problem_

Put the model into slic3r and slice.
* _Expected Results_

The bottom layer after support interface layer should always be filled.
* _Actual Results_

The last support interface layer
[image]

First bottom layer, in this case, it was not filled at all. This the the problem I am facing.
[image]

At second bottom, the solid infill appears.
[image]

3rd bottom layer, also with normal solid infill.
[image]

Next, the normal infill. So we got 3 solid layer, one of them has problem.
[image]

#### STL/Config (.ZIP) where problem occurs
Somehow I can not upload ZIP here, I put it on my googleDrive
There are test STL and my slic3r setting
https://drive.google.com/open?id=0B_K93ABCQm2RRWswcjlYSkdaRGM

metalzse, 2016-08-04:
Hi, should I provide more information for this issue?

lordofhyphens, 2016-08-04:
I can't see any "gaps" in the 3d preview on 1.3.0-dev with your model and settings. Additionally, remember that the support layers are a different thickness than the print/infill layers (usually), so "empty" layers on the 2d preview may not be actually empty. 

Please check with the 3d preview.

closed 2016-08-04 by lordofhyphens