slic3r/Slic3r #3348: Replacing part of a stored AMF file puts it far away on the platter

vlast3k, 2016-05-30:
### Version
1.3.0_dev
From 21.May.2016

```
C:\develop\Slic3r>git describe --tag
1.2.9-252-g9e8022f

C:\develop\Slic3r>git rev-parse HEAD
9e8022f6f62b6f4c07b5846a765128b3609e53ef
```
### Operating system type + version
Windows 8.1 x64

### Behavior
1. Store a model consisting of multiple parts in an AMF file
2. Delete All from the plate
3. Import AMF file
4. Open the object, remove a part and then add the same part

As a result the part is added in a completely different position in the space.
Thus the functionality to export files as AMF together with the slicing overrides is a bit crippled

#### STL/Config (.ZIP) where problem occurs
[example.zip](https://github.com/alexrj/Slic3r/files/289462/example.zip)

Example contains 2 stl files.
1. Import cube1.stl
2. Open it in the parts editor
3. add Cube2.stl as a part (both should be sitting next to each other)
4. save as AMF
5. follow procedure from the previous section
6. cube2 is imported far away from cube1

vlast3k, 2016-05-30:
Since i have the problem that i am slicing a model consisting of 5 parts with different settings, i finally ended up creating a tool, that takes an AMF file and a JSON file with the configuration and patches the AMF file with the overrides. It is located here, in case someone would use it

https://github.com/vlast3k/Slic3rAMFPatcher

While building the model i frequently modify parts and since Slic3r tends to crash from time to time, each time i loose my configuration and need to redo it.
With this tool the idea is to use Slic3r to make an AMF file, and then just patch it with my settings

lordofhyphens, 2016-05-30:
Yeah I have noticed that importing AMF of a plate auto splits it, STL
remains together. Probably an artifact of the file formats.

alexrj, 2017-03-17:
@vlast3k, thank you for this bug report. I fixed this. The original part coordinates are now preserved when exporting an AMF file.

closed 2017-03-17 by alexrj