slic3r/Slic3r #3350: Feature Request: Sequential Extruder changes

james-lan, 2016-06-01:
### Version
1.2.9 (and I think a later git version, but I'm not at the computer)

### Operating system type + version
Linux Kubuntu 15.10, 16.04, Windows 8.1

### Behavior
* _Describe the problem_
Using a color mixing nozzle, such as the Diamond Hotend, where the mixing is represented by multiple 'extruders' (handled by firmware), when sent to slic3r, slic3r will generate paths which seem to use an arbitrary assignment of which extruder to use first. This means that a mixing tower is needed when doing items with small amounts, in the case of a rainbow type item, see picture, while there would be a slight initial discoloration, the color change is not drastic between any of the particular 'extruders'. 

Without this, it needs a time and filament consuming purge tower, otherwise you can have blue in your orange, with nasty looking results. 

In the example picture, you can see most of them do change sequentially, and a purge tower is mostly a waste. I can play around with it a little to cause different ones to be ordered, by changing the order the stl files are loaded into repetier-host (screenshot is from there, as it is easiest to see, and an approximation of reality. Slic3r is not showing the travel moves.) As you can see it goes from the Blue (#2) to the green-green-blue (#11), then to green(10), to cyan(0), to blue-blue-green (1) then to yellow-green(9), and on down to like I think it should. (Actually, the order may be reversed, but you get the idea.) 
[image]
[image]

* _Expected Results_

An option to have the extruder change by only 1 at a time if there will be data on the line. (possibly with a looping option, between in this case 11 and 0 (or 12 and 1 depending on how you look at it)) 

As it exists, I can manually change the output, but that would require a lot of work. (So far, I don't trust any scripts to do it.) 

As a possible continuation of the basic request, and expansion, have a preferred change option for the extruders, to allow setting which extruders you would want one to change to by preference. (eg, in this example everything would be X+1 and X-1 except 0 and 11, which would have them.) 

_Is this a new feature request?_
Yes

I tried to look through the source to see about changing it myself to submit a patch, but wasn't able the code controlling extruder changes easily. 

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Uploading a copy of the exported config (export_config), and the overlay that repetier was using. (It really doesn't matter much, I've setup a clean slic3r copy and had the same results, adding in objects in different orders as well when creating the amf.) 

[slic3r-report.zip](https://github.com/alexrj/Slic3r/files/294131/slic3r-report.zip)

lordofhyphens, 2016-06-02:
I suspect finding the appropriate order may be computationally expensive (it's probably buried in the ordering of the loops calculated). It's a decent idea though.

james-lan, 2016-06-03:
My current plan is: A postprocessing script to adjust the order of the materials per Z layer, though I'm not very familiar with perl, but I don't think that's ideal. As it's simply in the X-Y plane, and not extruding the movements can be reordered.

bubnikv, 2016-10-27:
The place in the code where the extruders are ordered is very localized, so even a rookie programmer could tweek it. Look for         

        # tweak extruder ordering to save toolchanges
        my @extruders = sort keys %by_extruder;
...

in GCode.pm

Currently the order is fixed: it increases with the extruder index, with a single exception: The last used extruder at a layer will be the first used at layer+1.

arrickg, 2018-05-14:
@bubnikv regarding the exception you mentioned...

> _exception: The last used extruder at a layer will be the first used at layer+1_

What is the exact change one would need to make to GCode.pm to eliminate this exception? I am aiming to reset the order each time, so that the first extruder used at any given layer is always that with the lowest assignment.

would commenting out the following if-statement do the trick?

```            
            if (exists $by_extruder{$last_extruder_id}) {
                @extruders = (
                    $last_extruder_id,
                    grep $_ != $last_extruder_id, @extruders,
                );
```

bubnikv, 2018-05-15:
@arrickg I think so, commenting out this block should do the trick for you.

arrickg, 2018-05-15:
@bubnikv, yes, that worked! 

Thanks so much; I've been looking to fix this for a while.

TLAS11, 2020-01-14:
I think this change on the c++ library is the following in PrintGCode.cpp:

```
// tweak extruder ordering to save toolchanges
	/*
        auto last_extruder = _gcodegen.writer.extruder()->id;
        if (by_extruder.count(last_extruder)) {
            for(auto &island : by_extruder[last_extruder]) {
               if (_print.config.infill_first()) {
                    gcode += this->_extrude_infill(std::get<1>(island.second));
                    gcode += this->_extrude_perimeters(std::get<0>(island.second));
                } else {
                    gcode += this->_extrude_perimeters(std::get<0>(island.second));
                    gcode += this->_extrude_infill(std::get<1>(island.second));
                }
            }
        }
	*/
        for(auto &pair : by_extruder) {
            //if(pair.first == last_extruder)continue;
            gcode += _gcodegen.set_extruder(pair.first);
            for(auto &island : pair.second) {
               if (_print.config.infill_first()) {
                    gcode += this->_extrude_infill(std::get<1>(island.second));
                    gcode += this->_extrude_perimeters(std::get<0>(island.second));
                } else {
                    gcode += this->_extrude_perimeters(std::get<0>(island.second));
                    gcode += this->_extrude_infill(std::get<1>(island.second));
                }
            }
        }
```

Is there anyone more knowledgeable that can confirm?