slic3r/Slic3r #3352: Ooze prevention crash

VanessaE, 2016-06-02:
### Version
Slic3r git commit 9e8022f6f62b6f4c07b5846a765128b3609e53ef

### Operating system type + version
Linux (Debian)

### Behavior

Slic3r crashes when a large enough model is loaded and then exported to G-code, if multiple-extruder ooze prevention is turned on (Print Settings -> Multiple Extruders).

Watching the status at the bottom of the window, it goes through these steps, in order (I have background processing disabled):

* processing triangulated mesh
* generating perimeters
* processing triangulated mesh (again)
* generating perimeters (again)
* preparing infill
* infilling layers
* preparing infill (again)
* infilling laters (again)
* generating support material
* generating skirt
... and then it crashes before it starts writing the g-code file.

Smaller models such as the popular dual extruder calibration squares object on Thingiverse (http://www.thingiverse.com/thing:124450) make it all the way through the process and save out a working g-code file.

#### STL/Config (.ZIP) where problem occurs
Config bundle:  [Prusa i3 - Slic3r 1.3.0-dev config bundle.ini.zip](https://github.com/alexrj/Slic3r/files/295715/Prusa.i3.-.Slic3r.1.3.0-dev.config.bundle.ini.zip)
Use "Dual Material (two colors)..." for print settings, and "Dual extruder, increased retract" for printer settings.  I suppose any filament willdo, I was using my red and orange Inland PLA profiles.
Model (AMF): [aleph-2-color.zip](https://github.com/alexrj/Slic3r/files/295720/aleph-2-color.zip)

alexrj, 2016-12-22:
Thank you for the report. I think this was fixed.
@VanessaE, do you have a chance to test again?

VanessaE, 2016-12-25:
I'm afraid I do not, as I've returned my machine back to a single nozzle configuration.

alexrj, 2017-03-05:
Okay, closing then :)

closed 2017-03-05 by alexrj