slic3r/Slic3r #3357: Trying to change STL for bed shape leads to crash

VanessaE, 2016-06-03:
### Version

Slic3r from git, commit 9e8022f6f62b6f4c07b5846a765128b3609e53ef

### Operating system type + version

Linux (Debian)

### Behavior

* Go into Slic3r's Printer Settings -> General -> Bed Shape -> `Set...`
* Set it for custom/based on STL and load in an appropriate STL file.
* Save your config and exit from Slic3r.
* Run Slic3r again, go back to the Printer Settings/General section and click the Bed shape `Set...` button again.  Slic3r will crash with the following message:
```
Error in function BoundingBoxBase at src/libslic3r/BoundingBox.cpp:9: Empty point set supplied to BoundingBoxBase constructor
	 at /home/vanessa/RepRap/Slic3r-git/lib/Slic3r/GUI/2DBed.pm line 55.
	Slic3r::GUI::2DBed::_repaint(Slic3r::GUI::2DBed=HASH(0x4b929b0), Wx::PaintEvent=SCALAR(0x4b9ed10)) called at /home/vanessa/RepRap/Slic3r-git/lib/Slic3r/GUI/Tab.pm line 1019
	eval {...} called at /home/vanessa/RepRap/Slic3r-git/lib/Slic3r/GUI/Tab.pm line 1019
	Slic3r::GUI::Tab::Printer::__ANON__(Slic3r::GUI::Tab::Printer=HASH(0x447de28), Wx::CommandEvent=SCALAR(0x494e758)) called at ./slic3r.pl line 117
	eval {...} called at ./slic3r.pl line 117
```

#### STL/Config (.ZIP) where problem occurs
[prusa-i3-bed.stl.zip](https://github.com/alexrj/Slic3r/files/296901/prusa-i3-bed.stl.zip)

bubnikv, 2016-06-03:
Yeah, it is broken.

bubnikv, 2016-06-03:
I will look into it.

bubnikv, 2016-06-03:
Please cherry pick e0b986538669afe395afd3e37101f121753e73b0 from prusa3d.
Thanks, Vojtech

lordofhyphens, 2016-06-04:
@bubnikv 
`git checkout master; git checkout -b fix-3357; git cherry-pick e0b986538669afe395afd3e37101f121753e73b0` are the command to push that from your copy of our master branch

PR #3417 (2016-07-20): Fix of a crash when the Print Bed dialog is reopened after the bed shape
was defined with an STL.
Fix of rendering on Windows, refresh on resize.

Fixes #3357
Cherrypicked from prusa3d, thanks to @bubnikv
closed 2016-07-20 by alexrj