slic3r/Slic3r #3359: Cut mesh dialog does not work on Windows

bubnikv, 2016-06-03:
### Version
1.3.0-dev

### Operating system type + version
Windows

### Behavior
The cut mesh dialog is very slow: wxWidgets trigger a screen refresh many times per move of the slice plane slider.

When "cut" is clicked, nothing happens on Windows. This is an effect of a workaround, which Alex applied apparently for an old wxWidgets on Linux. After dialog EndModal(ok), calling Close() sets the return flag to "cancel". Depending on the platform, either ok (MacOS) or cancel (Windows) is returned.
The correct solution is to remove the Close() call.

Vojtech

bubnikv, 2016-06-03:
Please cherry-pick 62b440ff6c454099d27327c0f64b6c3ddd15eb72 from prusa3d branch.
Thanks, Vojtech

lordofhyphens, 2016-06-04:
@bubnikv the command line syntax to create a branch and cherry-pick this commit is:

`git checkout master; git checkout -b fix-3359; git cherry-pick 62b440ff6c454099d27327c0f64b6c3ddd15eb72`

Push the resulting branch to your fork and make a PR from that. Easy ;) You can delete it after the PR is merged.

lordofhyphens, 2016-07-06:
Has anyone else noticed this as a problem on Windows?

closed 2016-07-07 by lordofhyphens

fuding, 2016-07-09:
There are mistakes in the fix , operating system Windows.

`C:\Slic3r-master>perl slic3r.pl
Deep recursion on anonymous subroutine at C:/Slic3r-master/lib/Slic3r/GUI/Option
sGroup/Field.pm line 576.
Deep recursion on subroutine "Slic3r::GUI::OptionsGroup::Field::Slider::set_valu
e" at C:/Slic3r-master/lib/Slic3r/GUI/OptionsGroup/Field.pm line 551.
Deep recursion on subroutine "Slic3r::GUI::OptionsGroup::Field::Slider::_update_
textctrl" at C:/Slic3r-master/lib/Slic3r/GUI/OptionsGroup/Field.pm line 565.`

PR #3400 (2016-07-07): Performance fixes for plater cut.
Fixes a performance issues on Windows, where the cut is being
refreshed many times for a single move of the Z plane.

Fixes a problem on Windows, where the new wxWidgets always returned
Cancel, even if the window was closed with the "Cut" button.

Improved performance if the live preview is disabled or not needed,
because both top and bottom parts shall be displayed.

Closes #3359

Originally proposed by @bubnikv