slic3r/Slic3r #3361: First layer infill over-extrusion.

pholmes2012, 2016-06-04:
### Version
1.2.9
### Operating system type + version
Windows 10
### Behavior
First layer infill massively over-extrudes.
Load the attached STL file, slice it, print it.
#### STL/Config (.ZIP) where problem occurs
[bug.zip](https://github.com/alexrj/Slic3r/files/298899/bug.zip)

Mikk36, 2016-06-04:
Sample gcode file where the issue occurs?

pholmes2012, 2016-06-04:
Added zip containing the gcode.

Mikk36, 2016-06-04:
Doesn't seem to be an issue with current master branch, but I can see that it clearly is buggy in your provided gcode file.
Could you test it with the master branch (if you can compile it) or with this one: http://www.prusa3d.com/downloads/dev/slic3r/Slic3r-1-3-0-dev-win-5-5.zip ?

pholmes2012, 2016-06-04:
Sure, i'll give it a try if i can build it.

pholmes2012, 2016-06-04:
Ok, I tested it and the 1.3.0-dev build does not exhibit this problem.

closed 2016-06-04 by lordofhyphens