slic3r/Slic3r #3362: High DPI screen not supported

EiNSTeiN-, 2016-06-04:
I have a Surface Book, which has "high dpi" screen (224 dpi instead of the usual 96). Due to Slic3r using Wx, which depends on gtk2, the entire UI is incorrectly sized:

[image]

[image]

All the icons are too small, text and buttons are incorrectly sized and sometimes hard to read. This is a widespread problem for apps using gtk2 on hidpi screens, and unfortunately the only fix seems to be either upgrading to gtk3, or adding application code to scale everything back to proper size depending on the screen's DPI.

I am running Ubuntu 16.04 and just tested commit 223065221888e202a762bfeaff125f960896b009

bubnikv, 2016-06-04:
Hello Francois.

Slic3r uses the GTK library indirectly through the wxWidgets library.
You may try to compile wxWidgets against GTK3, see
http://docs.wxwidgets.org/trunk/page_port.html
"Support for GTK+ 3 is available starting with wxWidgets 2.9.4, use configure option –with-gtk=3 to enable it."

I suppose if you compile wxWidgets manually, then the Alien::Wx will find it. Then you may continue with the usual slic3r installation.

I am interested whether it will work for you.

Vojtech

EiNSTeiN-, 2016-06-04:
It seems there is no package or ppa for ubuntu xenial with wxWidgets 2.9.4, I will try building and installing from source and see if that fixes the problem.

bubnikv, 2016-06-04:
At prusa3d, we are building all the builds with the wxWidgets 3.0.2.
Currently we are testing the wxWidgets 3.0.3-beta, as these wxWidgets allow us to enable multi sample anti aliasing. This makes a great difference for the 3d layer preview.
Vojtech

sanchosk, 2016-09-14:
Hello, Vojto.
Can you, plz, describe how to enable the anti aliasing (and therefore also how to compile the slic3r against newer wx)?
I was looking to github - I thought Prusa would share this as the company is very pro-open source, but I could not find it :( Thanks!

bubnikv, 2016-09-14:
It is certainly not a secret.

You will find a sequence of steps to compile Slic3r with Visual Studio 2003
here:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

You will find some patches there and a description, how to compile
wxWidgets, Alien::Wx with wxWidgets 3.0.3-beta and a patch for OpenGL
package, which adds support for multi sample anti aliasing. When this is
set up, just go ahead and compile Slic3r, the support is already there.

Vojtech

sanchosk, 2016-09-14:
Hmmm, thanks, but I am working on Ubuntu. I'll read it anyhow and try to implement it with similar steps...

bubnikv, 2016-09-14:
> Hmmm, thanks, but I am working on Ubuntu.

Good for you.

> I'll read it anyhow and try to implement it with similar steps...

It is really about getting the wxWidgets-3.0.3-beta compiled with perl and
then adding a couple of constants to the wx OpenGL package. Getting
wxWidgets compiled is a question of either patching the alien::wx package
to pull and compile the right package, or doing it manually. The patch for
wx OpenGL package is quite minimal.

It would be great, if you could push the patches into the official
distribution of the alien::wx and wx OpenGl packages.

Good luck,
Vojtech

TheCodeSharman, 2017-07-28:
I've also got a HiDDPI screen - so would like to resolve this.

I was able to rebuild wxWidgets with:
` ./configure --with-gtk=3`
and then installed Alien::wxWidgets which correctly picked up the custom wxWidgets library as suggested by bubnikv above - this works fine. 

Almost. 

Now when I run slic3r everything is correctly sized except the GLCanvas widget which renders at half it's correct size:
[image]

The problem is that the OpenGL code in 3dScene is working in physical pixels whereas the GTK3 version of Wx is working in logical pixels. I'll look into patching Wx::GLCanvas to fix this.

bubnikv, 2017-07-28:
First you are reporting an issue with the Prusa3D fork. You should use the
github.com/prusa3d/Slic3r repository.
Second, this is a very old release, the problem has been fixed already.

TheCodeSharman, 2017-07-28:
ok thanks - I'll try with a later version