slic3r/Slic3r #3364: External perimeter value used instead of perimeter value

Tinchus2009, 2016-06-07:
### Version
_Version of Slic3r used goes here_
1.3.0.-dev

### Operating system type + version
Win10

### Behavior

Im useing 1.3.0-dev for about 3 months. It works GREAT. 2 days ago I received a new filament roll, PLA, a brand I never used. So I starter to do am extrusion calibration on it. As usual, I used the stl (atached). To my surprise I was way too underextrudeing. Wich was strange because my prints so far didnt show that problem. In fact my "magic number" for extrusion is always between 0.96/0.94. This time the result I got for extrusion multiplier was 1.11 !!!!
I srted useing that value with this PLA.... And I got obvious overtruded results.
I repeated the calibration:
I printed the STL with extrusion multipier value of 1, measured the walls and I got 0.42mm
Looking at the generated G-Code, the perimeter widht was set to 0.48, and external perimeters were set to 0.40. So, following the calibratio instructions, my STL walls were suposed to have 0.48, and my measured walls were 0.42... Again, my ersult said I was underextrudeing...

I repeated the same calibration process but useing slic3r 1.2.9 stable. Same config, same STL, same everything 9in fact config and profiles are shared between these 2 versions on my setup). The geenrated Gcode said: perimeters 0,48, external perimters 0.4, exactly as the previous 1.3.0-dev generated g-code. And I printed with this new g-code another STL. This time, the walls were 0.50 mm !!! 

Meaning: 1.3.0DEV  is useing external perimeter value for printing a single walled object, and in the previous version the value being used is "perimters", and at least if you follow the calibration instructions , that should be the correct value to be used.

Question is: is this a new behaviour of new slic3r versions or a bug?
I have atached STL and config 
[config.zip](https://github.com/alexrj/Slic3r/files/303632/config.zip)
[_40x20.zip](https://github.com/alexrj/Slic3r/files/303634/_40x20.zip)

Mikk36, 2016-06-07:
External perimeter -> outermost perimeter.
If there is only a single wall, it is the outermost perimeter.
This is working as intended.

Tinchus2009, 2016-06-07:
Ok... this is as it works now, it doesnt work like that in previous version. As I said, if you print the calibration object that I have atached, and if you follow the calibration instructions of the popular youtube videos out there, the value you should look at is the "perimeters" value. And that value is the one used in 1.2.9 and 1.2.7 . 
So Ok, I have discovered for myself that this is how it works now,  and it is ok, then it is not a bug, but cretainly it should be clarified for future calibration instructions

lordofhyphens, 2016-06-08:
Sounds like it was a bug that got fixed when @alexrj did the variable
perimeter width stuff.

Mikk36, 2016-06-08:
You can always manually override any extrusion width, so there should be no issues with calibration.

Tinchus2009, 2016-06-14:
No you cant. You can manually overrrida when the extrusion is 
calibrated. If you can not properly calibrate, how do you expect the 
machine will know how much to extrude?

Mikk36, 2016-06-15:
I meant that you can specify the exact extrusion width you want for any extrusion in Slic3r, so that you shouldn't have any issues calibrating your printer accordingly.
If you want a single-walled object with 0.5 mm extrusions, you can tell Slic3r to do that. Just set the external perimeter width to 0.5 and you'll get that.
You could also just set the default extrusion width to 0.5 and the rest to 0 and every extrusion width will be 0.5 then (except for bridging, that one is always assumed at nozzle diameter, because of physics).

Tinchus2009, 2016-06-17:
I understand what you say. You can set may extrusion to whatever you 
want, but, for example, if you specify an external wall to have 0.5... 
if the printer is not calibrated, there is no way to get those 0.5 as 
you want.

Anyway, the "problem" is not such a  problem, slic3r now is useing the 
external perimeter value instead of the perimeter value for a single 
wall object and that is all I need to now.

Thanks

lordofhyphens, 2016-06-18:
Alright, closing then.

closed 2016-06-18 by lordofhyphens