slic3r/Slic3r #3365: Outer perimeters of bridges not properly anchored

VanessaE, 2016-06-08:
### Version

git commit 223065221888e202a762bfeaff125f960896b009

### Operating system type + version

Debian

### Behavior

As it's drawing the perimeters of a bridge, Slic3r starts the thick part too far from the anchor point.  Instead, the bridge starts with a very tiny strand (a few tenths of a millimeter) at what looks like the normal extrusion rate for a perimeter, then it switches to the bridging flow ratio, creating a heavy strand with a weak anchor, which leads to the first layer of the bridge often breaking and drooping (at least in the case of PLA).  No amount of cooling or speed control has helped. 

This only seems to affect perimeters, especially external perimeters, where the difference between bridging flow and perimeter flow is greatest.  Solid infill seems to be fine in this regard.

[image]

#### STL/Config (.ZIP) where problem occurs

For a test model, use this bridging+retract/ooze test object, http://www.thingiverse.com/thing:15087 or you could use triffid_hunter's bridging torture test, http://www.thingiverse.com/thing:12925 (pictured above).  Any other model that invokes a normal bridging action should be fine too.
Config files don't matter here, as it's consistent across all of my print, filament, and printer profiles.

kyrreaa, 2016-08-22:
I've noticed this in 1.2.9a too. It even does a weird thing where it skips the columns of support...
[image]

bubnikv, 2017-01-27:
It is a good idea to improve anchoring of bridging perimeters. One has to be careful though about the appearance of the object's outer shell, if the "bridging perimeters" is used just for overhangs. Starting the bridging flow a bit earlier will likely cause marks on the object surface.

VanessaE, 2017-11-28:
What's the deal with this? Cam we please get a fix?

VanessaE, 2018-02-05:
This has been left to lie fallow for far too long, can someone PLEASE fix this?

lordofhyphens, 2018-02-06:
I am happy to support people trying to fix this; I have zero time myself to work on it.

It seems to me that it would occur less often with thicker perimeters, basically reducing the flow disparity between the two types of extrusion.