slic3r/Slic3r #3366: Feature request: Fan slow (gradual) start after first layer

bssstudio, 2016-06-09:
I would like to have the fan gradually gain speed when starting the second layer.
The motivation for that is the fact that turning the fan on can cause a significant drop in hotend temperature. When printing difficult materials, this can cause a jam or under-extrusion.
It would be nice to gradually ramp up the fan speed so that the hotent control loop can compensate for the additional power requirement caused by the part cooling fan.

lordofhyphens, 2016-06-10:
Are you sure that your fan shroud is designed and/or positioned properly? I
would not think it was supposed to be able to do that.

I had a post processing script for fan kickstart that may be adapted to
this.

mrvn, 2016-06-14:
I had my firmware report thermal runaway and lock up when the fan kicked in because of the temperature drop. Very annoying.

Mikk36, 2016-06-18:
A better solution is to insulate your hotend then.

bubnikv, 2016-06-18:
Maybe your firmware needs an adjustment to make the temperature feedback loop faster. Another solution, again on the firmware side, is to bump the speed of the temp feedback loop when the firmware kicks the fan on or off. There is really a lot, which may be done at the firmware side.

mrvn, 2016-06-20:
All good ideas. All harder, sometimes impossible, too do than being nice in the gcode. As for speeding up the feedback loop that partly depends on the hardware. There is a delay between the nozzle cooling down and the temperature sensor noticing given the distance between the two. Can't speed up the feedback loop faster than that, probably needs to be a magnitude slower to work.

VanessaE, 2018-03-04:
Something I would like to see is a gradual speed-up over some height.  The user could, for example, tell Slic3r to start off at 10% speed on layer 5, and gradually increase to 50% speed over the course of the next 20 or so layers, staying there for the rest of the print (unless layer time or bridging demands a higher speed, as usual).

This wouldn't be to avoid excessive hotend cooling (because PID should handle that), but to avoid excessive *bed* cooling.  Most heated beds are slow to respond to a temperature drop, no matter how well-tuned they are, and too much cooling in the first layers can cause lifting/warp.

f11h, 2019-01-21:
Any new on this topic?

lordofhyphens, 2020-07-27:
@f11h Nobody  has decided to take it up