slic3r/Slic3r #3367: Small Gaps in sliced object

Leviani, 2016-06-17:
### Version
_1.2.9_

### Operating system type + version
_Windows 10_

### Behavior
I had this issue with several stl Files, but not all of them. I can`t really find any reason why this happens. I tried to slice with 0.2 Layer height and 0.1 height and 3-5 Perimeter.
I also tried to disable and enable “Detect thin walls” and nothing seems to change the sliced result.

After printing the object has a lot of small riffles and cracks where it can be easily broken.
When taking a closer look at the sliced result and showing only a single layer I can see those little gaps, which seem quite randomly distributed around the object.
[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped co
[slic3r_settings.zip](https://github.com/alexrj/Slic3r/files/321016/slic3r_settings.zip)
py of an STL and your config (`File -> Export Config`)_

Mikk36, 2016-06-17:
An extrusion has to start somewhere, doesn't it?

Leviani, 2016-06-17:
@Mikk36 than the display of those start points changed. I could remember that in slic3r version 1.1.7 they were not displayed as such gaps. Maybe it has nothing to do with the start-gaps, but every time my printer produces prints that can be broken in half with no force I can find matching gaps in the slice.

Mikk36, 2016-06-17:
If that's so, then you should propably increase the temperature and maybe it's underextruding?

Leviani, 2016-06-17:
I think after 10+ successful prints with the same filament roll, the temperature should be ok. I just started another print with a cura sliced file and it doesn`t show any of those symptoms and the object looks just fine. I used the exact same temperature, filament diameter and flow settings as in slic3r (screenshot from same object, sliced with cura).
It seems like I am missing something else in the slic3r settings :(
[image]

Mikk36, 2016-06-17:
Have you specified a constant extrusion width the same as the nozzle width in Slic3r?
Cura always uses the same width as the nozzle width, which means that if you're running different widths in Slic3r, but same speeds, then you're actually printing slightly colder thanks to higher volumetric speed.

Leviani, 2016-06-17:
@Mikk36 thx, that setting broke my prints a few months ago ;)
So yes, since than I specified my Nozzle width to 0.3 both in cura and in slic3r. I also found the tip with the detect thin walls settings here https://github.com/alexrj/Slic3r/issues/2515 and tried to play around a bit with that but it didn’t help.

Leviani, 2016-06-17:
Ok, here the different results, the broken is sliced with slicer (settings in my first post) the other one was sliced with cura, same temperature, nozzle diameter, filament diameter settings. What settings do I miss?
[image]

Mikk36, 2016-06-17:
There definitely seems some underextrusion going on, see the missing external perimeter lines on the vertical wall.

Mikk36, 2016-06-17:
Try setting the default extrusion width to the as your actual nozzle width in Print -> Advanced and set everything else to 0.

Leviani, 2016-06-17:
@Mikk36 if you want to take a look in my ini settings (attached to my first post), my post 2 hours ago or this screenshot, than you will see that this is already the setting i used. And before you ask, yes I have mounted a e3d and changed the default nozzle with a new 0.3 nozzle and am quite sure that 0.3 is really the size of my nozzle ;)
[image]

bubnikv, 2016-06-18:
The zeros mean default value, which is the value which Slic3r finds the best for the purpose. The default value is not the default extrusion width.
If you want to print everything with 0.3mm extrusion width, you have to manually copy 0.3 to al the fields.

Leviani, 2016-06-19:
@bubnikv thx for that tip, I thought that the default extrusion width sets the "default" and as long as I leave the rest to 0 it takes the default value.
I tried another print with all Extrusion width values set to 0.3, with the same results :(
[image]

The interesting fact is, that only the small part seems to have those gaps, the larger part at the bottom has a seamless texture. 
[image]

And again, with the cura slicing (same temperature, same filament diameter, same flow/extrusion multiplier, same filament) the result doesn`t have that issues. Is there another setting in slic3r 1.2.9 that I am missing?
[standV2.zip](https://github.com/alexrj/Slic3r/files/322237/standV2.zip)

lordofhyphens, 2016-06-19:
Please provide a screenshot of Slic3r's preview, not RepetierHost.

Leviani, 2016-06-19:
@lordofhyphens you gave me an interesting idea, so I sliced the object again from the slicer UI and saved to an usb stick.
After comparing both gcode files (from slicer under Repetier Host and from the slicer UI) I found more differences than similar lines (what is find quite interesting).
Sadly, the print result is the same :(

Is there any option to zoom into the “Layers” View in the slicer UI? I can only see pictures of single layers that are really small.
[image]

Have you ever heard of those “gap” problems or similar phenomes before?

bubnikv, 2016-06-19:
The gaps you see in the visualization in the Repetier software are perfectly normal. Slic3r applies a hard-coded value to leave a tiny gap in the perimeters. These gaps are meant to hide the seams. Indeed, if the print head draws a perimeter, the start and end points don't end with a rectangular shape, but something circular. If the circular ends are squashed together, they will force the material outside the surface uncontrollably. The tiny gap is meant to suppress this effect.

Another gap I see from your visualization is at the starts of the infill lines. It seems the infill lines may not be sufficiently fused with the inner perimeter lines.

I believe your problem is somewhere else. My experience is, if you really set the parameters equally, Slic3r and Cura print similarly. Some of the parameters have no equivalent in the other software, but the general rule applies. It may be for example, that your Slic3r draws the infill much faster than the perimeters, thus causing insufficient nozzle pressure at the start of the infill lines and too much pressure at the start of a perimeter, causing blobs. It may be, that your Slic3r retracts too quickly, causing a nozzle jam. Hard to know.

Vojtech

lordofhyphens, 2016-06-19:
I meant the 3d preview in the slicer 1.2.9 UI. RepHost can't really work
out the trace width so it uses a uniform value. Slic3r's own 3d preview
does know about what the traces are supposed to look like.

Mikk36, 2016-06-20:
You do not have to put the same value to every field, if you've entered the default extrusion width, that's the whole point of it.
If you leave the default as 0, it will autocompute every other value, that's not already defined by you.

[image]
```
; generated by Slic3r 1.3.0-dev on 2016-06-20 at 11:59:17

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.40mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.40mm
```

[image]
```
; generated by Slic3r 1.3.0-dev on 2016-06-20 at 11:59:30

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.67mm
; infill extrusion width = 0.67mm
; solid infill extrusion width = 0.67mm
; top infill extrusion width = 0.67mm
```