slic3r/Slic3r #3368: Crash bug, Can't call method "print_z" on an undefined value at /usr/share/perl5/Slic3r/GUI/3DScene.pm line 1104.

lordofhyphens, 2016-06-18:
### Version
72980c2c3b5af7645c6d69d65df20a20b9cdedc3 and 5f521b24c42ed657967b919871900fa6a65ba790

### Operating system type + version
Debian Linux 8

### Behavior
* _Steps needed to reproduce the problem_
Add one of the head.amf and one of the hands.stl to the platter. 
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
Slic3r does not crash.
* _Actual Results_
Slic3r crashes, error:
Can't call method "print_z" on an undefined value at /usr/share/perl5/Slic3r/GUI/3DScene.pm line 1104.

#### STL/Config (.ZIP) where problem occurs
http://www.britishideas.com/8x2n/config.ini
http://www.britishideas.com/8x2n/Head.amf
http://www.britishideas.com/8x2n/Hands.stl
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-06-18:
Does not crash when each is exported separately.

alexrj, 2017-03-05:
This is fixed as of current master.

closed 2017-03-05 by alexrj