slic3r/Slic3r #3369: Feature Request: Ability to change config directory path

Flole998, 2016-06-18:
It would be helpful to change the config path for example to support network drives and keep the slic3r config files synced between all my computers. There should be a textfield in the preferences dialog to select a path where slic3r looks for it's configuration. If the config file at the default location contains such a location directive, slic3r should follow it and read the config file at that path. If any error occured it should warn the user and fall back to the default config file. Also there should be a check to prevent endless loops from one config to another

Mikk36, 2016-06-18:
Symbolic links can also be a solution. All operating systems support that nowadays.

lordofhyphens, 2016-06-22:
Also, trying to make it configurable from within Slic3r runs into a circular logic unless you are very explicit about what you want to be configurable. At most the program's self-configuration can't be altered (as where would it write it down)? 

It's certainly feasible to make the print, printer, and filament folder locations configurable though.

Flole998, 2016-06-22:
I was talking about print, printer and Filament configurations. The Problem i have with symlinks, that if the server goes down or is unreachable, nothing would work. Also using an option in the settings Dialog it would be possible to quickly change that path, even for users who might not have access to symlinks or the App data Folder as per group policies in a company environment running Windows.

Mikk36, 2016-06-22:
I was more thinking of symlinking to another folder, which is synced to a "cloud" or similar, not directly to a networked folder.
That way if You update settings in one computer, they get pushed to the others when network/server is up. If not, you still have your local files.

Flole998, 2016-06-22:
No normal User is going to open cmd and enter a mklink command on Windows to create a link. Yes, there are other ways than adding this feature, but most if them are dirty and require an untouched Operation System. However, most companies using Windows are restricting the use of cmd and also block access to certain Folders.

vlast3k, 2016-06-23:
> We'll take a pull request for it (it has to work on all 3 platforms slic3r
> is on), but that's all I can say about adding this feature.

and what about the --datadir command line parameter, doesn't it help ?

alexrj, 2016-07-10:
Yes, @Flole998 why don't you use the --datadir option?  You can easily make a .bat file that launches Slic3r with the --datadir option.

closed 2017-02-04 by Flole998