slic3r/Slic3r #3370: "zick-zack" line on perimeter infill!?

Hecot, 2016-06-20:
Usually I have a lot off thin walls in my prints. When I choose 3 perimeters on the outside and the wall is thinner to let 3 perimeters fit in the inner perimeter will be replaced by a small "zick-zack" line:

[image]

But when this "zick-zack" line is printed my printer **always** over extrude and leave a big hump in this area and mess the print!!! Is there any way to control this "zick-zack" line by an separete extrusion multiplyer or at least a way to switch this automatic replacement off!?

lordofhyphens, 2016-06-20:
Please provide a STL and config.ini where you are seeing this problem, and provide a screenshot from Slic3r's 3d preview, not repetierhost.

lordofhyphens, 2016-06-20:
Also, try 1.3.0-dev (there's a build available www.prusa3d.com/downloads/dev/slic3r )

bubnikv, 2016-06-21:
Where does it over extrude? Over the full layer, or just where the infill lines flip their direction?
You may try to set the infill line width explicitely to the same value as the nozzle diameter. Slic3r by default uses a much wider line width for the infill than for the perimeter, which may cause problems with some extruders I suppose, if the nozzle does not iron out the material sufficiently flat.

Hecot, 2016-06-21:
Here are the information - when you look at the height of 19.80 mm you can see that there is only 2 Perimeters and not three as in the lower part. 

[image]

These small "zick-zack" infills causing trouble as there is too much plastics extruded in this area, unfortunatly I have no print but to show you the problem.

[zick-zack.zip](https://github.com/alexrj/Slic3r/files/326313/zick-zack.zip)

Hecot, 2016-06-21:
I found when I switch on "gap-fill" (was set to zero) the "zick-zack" infill lines are disappearing and will be replaced by a solid line! 

[image]

OK - this solved my problem but I still don't understand the logic that the gap will now be filled with infill. To me it would be more logical if there will be a gap as that is what I expect from switching gap-fill off! Maybe someone can think about this logic and change in future version - I close the issue!

closed 2016-06-21 by Hecot

lordofhyphens, 2016-06-21:
What your screenshot looked like to me was exactly how it was supposed to.
That your printer wasn't able to produce that isn't Slic3r's fault.

I would make sure your printer is not overextruding just in case.