slic3r/Slic3r #3371: Feature Request: different layer heights for different things

mrvn, 2016-06-23:
Currently slic3r uses different layer heights for the first layer, other layers, infill (combine infill feature) and support. With the push towards auto adjusting layer height the layer height becomes much more fluid and I think more control over it would be good.

I therefore request separate layer height settings for different things. For auto adjusting layer heights those should be min/max pairs: (from memory)

1. skirt
2. brim
3. outer perimeter
4. perimeter
5. infill
6. top/bottom fill
7. bridging
8. support
9. raft

The infill setting would replace the "combine infill" feature obviously. The extrusion width mask could be combined with layer height (min/max). Everything there should have height too and vice versa.

ldpgh, 2016-06-26:
> 3, outer perimeter

In order to get a smoother surface (PLA) I'd like to print  just  the most outer perimeter with a reduced layer thickness (e.g. 0.25mm standard layer height ... just most outer perimeter 0.05mm ... keep everything else as it is)

lordofhyphens, 2016-06-26:
- Skirt seems pointless, it isn't there as part of the object so it should be calculated to properly prime nozzles
- Support is pointless, for reasons I mentioned in #3377 .
- perimeters and infill need to be even multiples of each other, as well as top/bottom fill.

We don't have automatic adjustable layer heights yet, so please give concrete examples where things can go awry using the results from the PR that implements it.

alexrj, 2016-07-10:
Yes, as @lordofhyphens said there are too many conflicts in this. I invite you to try to design the logic for such feature and you'll understand it's almost not doable...

closed 2016-12-18 by alexrj