slic3r/Slic3r #3372: Feature Request: spacing parameter for infill

mrvn, 2016-06-23:
The recently closed #1644 made me think of this although I don't think it was what the original submitter ment. So opening a new issue.

When the top fill is printed it will sag between the infill strands. The larger the gap between infill the more sagging happens. Similar the wider the infill pattern the more flexible the sides of an object become, allowing the perimeter to bend when pushed. Now the spacing of the infill depends on 2 parameters: Infill percentage and extrusion width. And extrusion width depends on layer height. This makes the spacing of infill rather flexible.

Wouldn't it make sense to (optionally) specify a spacing (defined as the maximum distance between any 2 points where a line between them does not cross any infill)?

bubnikv, 2016-06-24:
Sounds like the following announcement:
http://hackaday.com/2016/06/10/3d-internal-structure-for-better-3d-printed-objects/

It may be useful to some, but it would add yet more clutter to the UI.

Vojtech

burdickjp, 2016-06-25:
I think it could be accomplished similarly to the way other things in Slic3r are: it's either a percentage or a dimension. When it's a percentage it behaves as it does now. When it's a dimension it behaves as described above.

lordofhyphens, 2016-06-28:
Isn't this a dupe of #3333 ?

burdickjp, 2016-06-28:
I don't think so, but I'm not the original poster.
I think we're asking for the ability to change from having infill be a calculated percentage to being able to dictate in dimensions the distance between infill features.