slic3r/Slic3r #3374: Feature request:  multiple temperature-vs.-speed settings in filament profile

VanessaE, 2016-06-24:
I mentioned this in a comment on another issue, and figured I ought to formalize it here.

When you set up a "Print Settings" profile for a given speed, one thing that you must do along with it is set up a corresponding "Filament Settings" profile with compatible temperatures (and perhaps cooling).  Faster speeds require more torque on the extruder motor, or hotter temperatures at the hotend.  The latter is easily solved in software, but the former, not so much. :smile:

This leads to multiple duplicates of each Filament Profile, which if you have to alter a setting su

I think what should be added is a feature in the Filament profile that will make it vary the temperature based on a range of speeds.  For example, currently Slic3r has:
```
Temperature (°C)
Extruder:  First Layer: [ 225 ] Other layers: [ 225 ]
Bed:       First Layer: [  60 ] Other layers: [  60 ]
```
What I propose is:
```
Temperature (°C)
    Extruder:  First Layer: [ 195 ] Others: [ 195 ] Speed: [   0 ] to [  59 ] mm/s
Bed:       First Layer: [  60 ] Other layers: [  60 ]
[X] Extruder:  First Layer: [ 205 ] Others: [ 205 ] Speed: [  60 ] to [  99 ] mm/s
Bed:       First Layer: [  60 ] Other layers: [  60 ]
[X] Extruder:  First Layer: [ 225 ] Others: [ 225 ] Speed: [ 100 ] to [ 199 ] mm/s
Bed:       First Layer: [  60 ] Other layers: [  60 ]
[X] Extruder:  First Layer: [ 260 ] Others: [ 260 ] Speed: [ 199 ] to [ 999 ] mm/s
[Add More...]
Bed:       First Layer: [  60 ] Other layers: [  60 ]
```
I don't propose to vary the print temperature in realtime. Rather, each entry would set the "baseline" print temperatures according to whatever is the *top* user-specified speed for the Print Settings profile that's about to be used for the print.  In this example, if the user's current Print Settings has their top speed at, say, 125 mm/sec, then only the third line would come into effect, for the entire print.

[Add More] would, of course, be a button to add another such rule.

I suggest a similar feature be added under the "Cooling" section of the Filament Settings.

VanessaE, 2017-05-20:
After some thought, I've come up with a better suggestion for this feature, if it should be added:  a simple formula that the user can plug a couple of values into, not too different from Marlin's auto temperature, but which would work on all firmwares and machines.  I've edited the opening post accordingly.

VanessaE, 2018-06-24:
It may also be prudent to include two multipliers - one for the flow-related change, and another for the overhang/height ratio.

VanessaE, 2018-06-27:
Over a year old now with no input from anyon , and nothing I can think of can really replace discrete filament+printer profiles tuned for each use case.

VanessaE, 2018-07-12:
Another idea:  what if in the filament profile, the user could just create a list of print profiles from those available, and assign custom temperatures to each (in addition to the "default" temperature)?  Instead of fancy math about flow rates and overhang and so on?

So e.g. gold v2 PLA might have 205° for default, but the user could select from a drop-down such that their "120 mms, 0.2 mm layers" profile would print at 215° with just that one filament, while another filament might have a default of 195°, but be able to rise to 200° for that same profile, etc.

This would not operate like the shortcuts we have now.  Rather, this would be embedded into and contained within the filament profile.  Each would have its own print-profile-to-temperature mappings independent of the others.

lordofhyphens, 2018-07-12:
@foreachthing any thoughts on this topic?

foreachthing, 2018-07-13:
I don't know! Sounds like a lot of work for something I can easly change in profiles. Maybe I don't understand the benefit of this adaptive temerature function. 
I'd rather have the temperatur setting as a shortcut - done! Since you already know (or need to know) the temp. per filament.... 

https://www.explainxkcd.com/wiki/images/e/e5/is_it_worth_the_time.png

VanessaE, 2018-07-13:
The problem is remembering dozens or hundreds of different combinations of filament and print settings.  What prints fine at 60 mm/s and 200° doesn't print as nice at 120 mm/s.  It's the computer's job to remember that.  MY job is to remember obscure Star Trek trivia (i.e. I have better things to occupy my mind :P smiley:)