slic3r/Slic3r #3375: Hi,

MartinMajewski, 2016-06-24:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

MartinMajewski, 2016-06-24:
Hi,

i had really hoped, that a new macOS release with a fresher perl install out of the box would help resolving issues during slic3r's build. But sadly, no!

brew install wxmac -> success
Build of slic3r's GUI still requires Alien::wxWidgets

cpan install Alien::wxWidgets -> (after some fiddling with force) success
Build of slic3r's GUI tries to install Wx -> failure

Installing Wx manually with cpan or cpanm in verbose mode tells me the following:

`DataView.c:3434:20: error: no matching constructor for initialization of
      'wxDataViewEvent'
      RETVAL = new wxDataViewEvent( type, id );
                   ^                ~~~~~~~~
/usr/local/include/wx-3.1/wx/dataview.h:809:5: note: candidate constructor not
      viable: requires single argument 'event', but 2 arguments were provided
    wxDataViewEvent(const wxDataViewEvent& event)
    ^
/usr/local/include/wx-3.1/wx/dataview.h:801:5: note: candidate constructor not
      viable: requires 3 arguments, but 2 were provided
    wxDataViewEvent(wxEventType evtType,
    ^
/usr/local/include/wx-3.1/wx/dataview.h:784:5: note: candidate constructor not
      viable: requires 0 arguments, but 2 were provided
    wxDataViewEvent()
    ^
/usr/local/include/wx-3.1/wx/dataview.h:791:5: note: candidate constructor not
      viable: requires at least 3 arguments, but 2 were provided
    wxDataViewEvent(wxEventType evtType,`

Since slic3r's last release is over one year old and has many bugs that makes it not suitable for me, I switched to Simplify3D. But now I am in a stage where even Simplify3D has limits that prevents me from getting things done. I thought that I could use slic3r again because of all the work done in the background. Can anyone help me, or at least provide me the latest nightly build of slic3r, please?

Bye
Martin

bubnikv, 2016-06-24:
Martin, you may try
https://github.com/prusa3d/Slic3r/releases
Vojtech

MartinMajewski, 2016-06-24:
Genius! Thank you very much!

Although being a very good application, slic3r is a pain in the butt when it comes to compiling it. I cannot reserve a Mac with exactly the dependency, os and tool versions that slic3r requires exclusively for it. :-(

lordofhyphens, 2016-06-24:
Yeah. I don't even have a mac to do builds on, so there's that

lordofhyphens, 2016-06-25:
Um, could you use Perl 5.20.x or something less ancient than 5.8 and see if that helps?

lordofhyphens, 2016-06-25:
@MartinMajewski I've cleaned up the thread a bit.

bubnikv, 2016-06-25:
We built Slic3r on MacOS using the latest perl from https://perlbrew.pl/ .
Vojtech

MartinMajewski, 2016-06-25:
Okay, I installed perlbrew, perl 5.24.0 and did a fresh clone on the Slic3r git repo.
Building Slic3r without the gui flag resulted in this failure report:

`Test Summary Report
-------------------
t/angles.t                   (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/arcs.t                     (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/avoid_crossing_perimeters.t (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 1 tests but ran 0.
t/bridges.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 16 tests but ran 0.
t/clean_polylines.t          (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/clipper.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/collinear.t                (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/combineinfill.t            (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/config.t                   (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 1 tests but ran 0.
t/cooling.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/custom_gcode.t             (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 15 tests but ran 0.
t/dynamic.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/fill.t                     (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/flow.t                     (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/gaps.t                     (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 1 tests but ran 0.
t/gcode.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 23 tests but ran 0.
t/geometry.t                 (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/layers.t                   (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 5 tests but ran 0.
t/loops.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/multi.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 13 tests but ran 0.
t/perimeters.t               (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 59 tests but ran 0.
t/polyclip.t                 (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/pressure.t                 (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 1 tests but ran 0.
t/print.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/retraction.t               (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 22 tests but ran 0.
t/shells.t                   (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 21 tests but ran 0.
t/skirt_brim.t               (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 6 tests but ran 0.
t/slice.t                    (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/support.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 27 tests but ran 0.
t/svg.t                      (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 2 tests but ran 0.
t/thin.t                     (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 23 tests but ran 0.
t/threads.t                  (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: No plan found in TAP output
t/vibrationlimit.t           (Wstat: 512 Tests: 0 Failed: 0)
  Non-zero exit status: 2
  Parse errors: Bad plan.  You planned 9 tests but ran 0.
Files=33, Tests=0,  1 wallclock secs ( 0.10 usr  0.08 sys +  0.81 cusr  0.22 csys =  1.21 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!`

lordofhyphens, 2016-06-25:
Failed tests means it still built.

MartinMajewski, 2016-06-25:
just wanted to report this to provide you guys some information.

Now I try the gui build...

MartinMajewski, 2016-06-25:
Result is again a failing WX build:

--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
Configuring Wx-0.9928 ... OK
Building and testing Wx-0.9928 ... FAIL
! Installing Wx failed. See /Users/martinmajewski/.cpanm/work/1466858352.89530/build.log for details. Retry with --force to force install it.
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
==> Found dependencies: Wx::build::MakeMaker
--> Working on Wx::build::MakeMaker
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
Configuring Wx-0.9928 ... OK
Building and testing Wx-0.9928 ... FAIL
! Installing Wx::build::MakeMaker failed. See /Users/martinmajewski/.cpanm/work/1466858372.90140/build.log for details. Retry with --force to force install it.
! Installing the dependencies failed: Module 'Wx::build::MakeMaker' is not installed
! Bailing out the installation for Wx-GLCanvas-0.09.
Don't worry, this module is optional.
The following prerequisites failed to install: Wx

bubnikv, 2016-06-25:
The repot says it all:
> See /Users/martinmajewski/.cpanm/work/1466858372.90140/build.log for
details.

MartinMajewski, 2016-06-25:
Thank you for pointing me to the log.

One mistake I made was not to install cpanm for perlbrew. It was still the system's cpanm.

After installing cpanm for perlbrew the Wx build process went much further, but stops then with the same error as I had before perlbrew with my "ancient" perl version. Log attached...

[WxFailLog.pdf](https://github.com/alexrj/Slic3r/files/333065/WxFailLog.pdf)

bubnikv, 2016-06-25:
Wx installation is always a PITA. I suppose the pain depends on the OS-X
version as well. It may be, that the Alien::Wx tries to use the default
wxWidgets installed on your OS-X. You may try to force Alien::Wx to
recompile and install its own WxWidgets into your local .cpan directory.
Vojtech

MartinMajewski, 2016-06-25:
Thank you for your reply @bubnikv 

One question: Which module do you mean with "Alien::Wx"? Alien::Wx does not exist as such. Could you please guide me a little more on how to implement your suggestion.

Thanks

Bye,
Martin

bubnikv, 2016-06-25:
it is Alien:wxWidgets.
http://search.cpan.org/~mdootson/Alien-wxWidgets-0.67/lib/Alien/wxWidgets.pm
Just read the manual and try to configure the build process to pull and
compile the latest wxWidgets, which is 3.0.2.
I don't remember the process step by step anymore, we have a build machine
configured and running, which is what counts.
Vojtech

MartinMajewski, 2016-06-25:
So I have more ideas what to do.

I got Alien::wxWidgets and Wx to build with cpan, but it always fails with cpanm

Can I somehow link to the working cpan installation of Wx?

bubnikv, 2016-06-27:
I posted the writeup on how to build a Slic3er distribution on MacOS and Windows to the Prusa3D fork of Slic3r
https://github.com/prusa3d/Slic3r/commit/3e8575c9315f8f61bb7ff38e149f7cf83c25bde1
Vojtech

lordofhyphens, 2016-06-28:
@bubnikv  https://github.com/prusa3d/Slic3r/blob/release_candidate_1_3/doc/How_to_build_Slic3r.txt 

Been meaning to ask, is the release_candidate_1_3 branch on prusa3d your default branch? Might be helpful to your customers to make sure that the default-displayed branch on github is the one with all of your patches on it, similar to how http://github.com/lordofhyphens/slic3r defaults to the debian branch.

lordofhyphens, 2016-06-28:
@bubnikv can I steal that document and wikify it and toss it up on the main slic3r wiki?

drgiller, 2016-07-28:
5.8.2?  Just a sanity-check here... Even OSX 10.11 ships with Perl **5.18.2**.  Was your 5.8.2 report a typo missing the 1, or did you use some other method to get such an old Perl?  OSX hasn't shipped with 5.8 for almost **10 years**.  It seems to me this issue should be closed.

MartinMajewski, 2016-08-19:
@drgiller It is a typo, an 1 is missing. macOS 10.12 comes shipped with 5.18.2

alexrj, 2017-03-17:
I'm closing this one as it's a bit old.
We now have automated OSX builds at http://dl.slic3r.org/dev/

Compiling under OSX 10.12 is still not easy because our dependencies have issues on that operating system version, but we're trying to sort everything out and document the procedure.

closed 2017-03-17 by alexrj