slic3r/Slic3r #3376: Support material z-distance: wrong variable used

Ricky1975, 2016-06-25:
### Version
1.3.0-dev
1.2.9-265-gcb72967

### Operating system type + version
Win7x64 (any)

### Behavior
Distance between support material and first object layer is calculated wrong: $nozzle_diameter is added but not usefull, if distance is set manually. Maybe I'm wrong, but please check this.

Current SupportMaterial.pm:
```perl
sub contact_distance {
    my ($self, $layer_height, $nozzle_diameter) = @_;
    
    my $extra = $self->object_config->support_material_contact_distance;
    if ($extra == 0) {
        return $layer_height;
    } else {
        return $nozzle_diameter + $extra;
    }
}
```

==> Maybe should be:
```perl
sub contact_distance {
    my ($self, $layer_height, $nozzle_diameter) = @_;
    
    my $extra = $self->object_config->support_material_contact_distance;
    if ($extra == 0) {
        return $layer_height;
    } else {
        return $extra;
    }
}
```
br
Hendrik

Ricky1975, 2016-06-25:
Also a question: Why is the nozzle width always used to calculate support layer heights? Why is there a hard coded factor of 0.75?
Wouldn't it be much more efficient, if there is a user controllable parameters to define "Print Support every x layers" affecting only the layers below the interface?

lordofhyphens, 2016-06-26:
Because of a general rule of thumb that you get poor adhesion/extrusion
with several layers  > 75% of nozzle width.

lordofhyphens, 2016-06-26:
Yeah, I'm not sure why it's doing this. Easy enough to test to see if it does anything weird though with the change :) Suggest running a few print tests with the variable changed (if you can run a build, it's trivial to change stuff like this and try again because you just need to restart Slic3r and not rebuild).

If anything the way it is biases at being slightly more distant from the previous layer. The distance used to be nozzle width * 1.5.

@alexrj any reason why this is the way it is? I see it's been there since the option was added.

Ricky1975, 2016-06-26:
@lordofhyphens 
Thx for fast reply, in regard of the rule of thumb. This is ok. I think, relying on the nozzle diameter as limiting factor is reasonable, but using the nozzle diameter for not user controllable calculations is worth thinking of.

In regard of the formula above to calculate the support layer distance I would suggest to change it to `min(layer_height*2,nozzle_diameter * 0.75)`. Because you should not assume a 0.75 * nozzle_diameter is always the best solution for every printer. (Or at least make the "security factor" 0.75 user configurable).

@alexrj & @lordofhyphens 
I implemented my change mentioned in the entry post in my local code and it works perfectly as I need a 0.15mm spacing for 0.1mm layers with a .4mm nozzle. In the original version I always ended with 0.4mm or more.
I think, this is just a bug not found for a long time, as a lot of people using slic3r try to use no support or rely on meshmixer as myself.

lordofhyphens, 2016-06-26:
Yeah, it seemed odd. What I can do is put together a PR for that change and
put it up for more widespread testing.

The other change to have it take the min of the layer height + extra for
non soluble support also seems like a fair plan, but it definitely is
something that would need testing. I will add it to the PR.

lordofhyphens, 2016-06-26:
Looking over the other change I think it needs some more experimenting or testing with different layer heights, so I'll put a separate PR in for it. The comments indicate that the max() is used to avoid inserting a ton of ultra-thin support layers; that is, it defaults to as thick as layers as it can manage for the non-contact areas.

Moreover, slic3r won't even let you choose layer heights > nozzle diameter to begin with. If you're having flow issues at 75% of the stated nozzle width I'm wondering if you're lying to slic3r about the nozzle width (in which case you deserve what you get ;) )

Ricky1975, 2016-06-26:
@lordofhyphens 
Of course, the adaption of layer heights need some testing, as it only is an idea. Yes, I have flow issues if using .75 * nozzle_diameter (without lying :-) ) I create technical designs and this depends on mechanical stiffness. I also do not care of printing time very much. Quality is the key for me (and therefore I switched so slic3r as nearly everything can be configured by hand).
So this is no issue of slic3r itself but an issue of my quality expectations - and as slic3r is the best tool with the most user tweak able settings, I suggest to make this factor tweak able, too. Maybe by just making the 0.75 security factor tweak able by the user for all calculations. I would prefer .5 for my personal usage.
Thx for adding the other fix - I hope it will find it's way into the official code soon :-)

lordofhyphens, 2016-06-28:
@Ricky1975 #3377.

There's a win32 package for that and other things at https://github.com/alexrj/Slic3r/files/336282/slic3r-most_prs.zip

lordofhyphens, 2016-06-29:
The PR needs a bit more work I think though; I can just see a bunch of users complaining that their support contact distance is too close now with this PR.

Ricky1975, 2016-06-30:
First of all - I don't see the complaints - please tell me where. 
On the other hand, of course this will happen if they do not change their values to reasonable settings (default). If they used something like 0.00001 before to minimize the distance (just keep extra above 0) and tweaked their nozzle size to .25 they got a distance of about 0.25mm before and will now get 0.00001 (extra) + 0.1mm (layerheight).
So everybody complaining should reset their values to reasonable settings (just an idea where this error might come from, because we should not assume, everybody understands the change but the setting are kept).

lordofhyphens, 2016-06-30:
I have learned to not assume that people are easy-going about changes is
all :)

The PR is going to get merged in regardless (well barring something
catastrophic coming of it) though and we will have to update documents and
the wiki.

I would appreciate it if people testing this did print tests too with it.

lordofhyphens, 2016-06-30:
I think what I am going to do for the default value is change it to 0.4; in
a lot of use cases this should yield similar results to 0.2 with the old
setup.

I will also probably change the ui to be a straight number in mm with a
tooltip stating that it is an offset from the previous layer, and that 0 is
soluable.

Probably could use some people doing some SCIENCE!! to work out some sane
defaults based on layer height.

Mikk36, 2016-06-30:
Did some prints today, worked just as expected. Finding a good value is an art though, but that's how it's supposed to be, there is no single best value.

lordofhyphens, 2016-07-11:
@alexrj I would appreciate your input on this problem. I think that the addition of the nozzle diameter to the extra when non-soluable does constitute a UI bug at minimum and causes failures at smaller layer heights.

The referenced PR removes the special casing, permitting support separation values 0> and <= nozzle_diameter.

Martronic-SA, 2016-09-15:
Ok I better understand why the support material doesn't follow my settings!
When I'm printing 0.1mm layer height with distance set to 0.1mm, the real support distance is always much greater and the upper part doesn't rely on the support.
If I set 0.1 mm, I was expecting to get 0.1mm even if it's not enough and my piece sticks to the support.

lordofhyphens, 2016-09-15:
Try the branch referenced on the pull request @martronic-sa

VanessaE, 2017-01-15:
There hasn't been much movement on this PR; would this also fix the case where the actual contact distance is affected by the angle of the supported face?

Seems like no matter what I do, I can't get a consistent contact distance between support and object, even at routine 0.1mm layer height.

bubnikv, 2017-01-15:
There are two types of contacts. Top and bottom. Which one are you
referring to?

I am currently tuning the new supports. I am getting a good top contact
layer (the one below the  overhangs), but the bottom contact layer is not
that easy to remove. I am printing the test objects side by side, sliced by
the Slic3r and simplify3d. The simplify3d has their supports tuned very
well. The pla supports remove in a single piece without leaving a scar. It
works like a magic. That is the benchmark I would like to reach.

VanessaE, 2017-01-15:
I was referring to top contacts.

bubnikv, 2017-01-15:
Just to make sure. You mean the contacts supporting overhangs. What
material? That makes a lot of difference.Pla cracks away easily, pet is
worse, abs is very difficult as the contact layer welds firmly no matter
what you do and how large the distance is, as long as the cooling fan is
off. With abs my impression is, you better use a sparse interface.

By the way, I developed couple of blisters on my hands over the last couple
of days just by printing multiple models and removing the supports myself.
It gives one quite an idea about the usability of the supports.

I think you requested a feature for soluble supports to print a non-soluble
interface layer below the first soluble layer. We are having the same
problem sticking PVA over pla with our multi material system, so I will
likely have to develop something.

VanessaE, 2017-01-15:
Yes, I meant exactly that.