slic3r/Slic3r #3377: Feature Request: combine support material every X layers

burdickjp, 2016-06-26:
Similar to how infill can be combined every so many layers, I'd like to combine support material every so many layers.
I have a build right now which would have taken over 50 hours with 0.1 mm layer height for everything. I decided to use Simplify3D for it, as it allowed me to set a 0.1mm layer height for perimeters, 0.2 mm layer height for infill and 0.3 mm layer height for support material. It will take around 30 hours, and I haven't thus far noticed any degradation in print quality.
If we were really clever we could keep the interface layers at regular layer height, too.

lordofhyphens, 2016-06-26:
Support material should already be done at maximum layer height possible.

burdickjp, 2016-06-26:
I don't think I've ever watched it behave any differently than printing support every layer.
I just loaded a model and went through it layer-by-layer. It skips every now and then, but not consistently. I'd like to be able to dictate this myself.

lordofhyphens, 2016-06-26:
Support in Slic3r should be always done at 75% of nozzle width for layer
height (#3376). I don't remember offhand how the interface layers are
generated, though.

I think the interface layers are scaled to provide the correct spacing, but
that's about it.

It would be fairly straightforward, I think, to hack the current combine
infill code into the support code, but if the layer height is already fixed
at maximum then there isn't a point to it.

lordofhyphens, 2016-06-26:
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/SupportMaterial.pm#L330 is the start of the method that generates the support layers and figures out how thick they need to be. It's either the object layer height or 75% of the nozzle width, whichever is larger.

Since you can't reliably print thicker layers than that there's no point in "combining" (that is, printing thicker) support layers.

If you really really want to experiment with different layer heights for support, you can do a source build (installing Strawberry Perl 5.22 and getting boost 1.59 and following the directions for building on windows works great on my win10 box), and just edit line 
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/SupportMaterial.pm#L340 and rerun slic3r.pl.

I've noticed that Slic3r's preview is not entirely accurate when it comes to combined infill, either, nor would most external gcode viewers anyway. 

If you have a specific test case where it doesn't produce support layers as stipulated above, please post the model and your config settings and we can figure out if it's actually a bug or if something else has gone wrong.

lordofhyphens, 2016-06-26:
If you're wondering about why 75%: 

- http://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide
- http://airwolf3d.com/question/layer-height/

burdickjp, 2016-06-26:
I appreciate the explanation and links and agree with why there is an upper limit to the extrusion volume. I still think this should be user-settable. If for no other reason than being able to manually vary it over Z.

I'd argue these same extrusion volume limits affect infill and yet we have manual control over skipped layers for infill.

lordofhyphens, 2016-06-26:
The assumption when using the skipped layers is that the combined infill
the layer thickness doesn't get larger than 75-80% of the maximum.

If you want to see how it works in practice with a fixed, larger, support
layer then change the line I mentioned earlier.

burdickjp, 2016-06-26:
I'm not necessarily wanting a larger support layer. In my example I have a 0.4 mm nozzle, 0.1 mm perimeter layers, 0.2 mm infill layers and 0.3 mm support layers. There are places on some of my models where finer layer control would be advantageous. Having support layer height be controllable allows it to be varied by model and varied in z.

lordofhyphens, 2016-06-26:
It's support. The point of support is that you throw it away when the print finishes. According to the way the source is written, what your example gave is already happening (0.4 *0.75 = 0.3), assuming you set combine-infill.

What I do know is going on currently is that for contact distance > 0, the air gap is nozzle width + whatever you specify (instead of what you specify over 0). That is definitely inconsistent behavior and there's a PR out to fix it.

burdickjp, 2016-06-26:
Support is important! Just because its job is short-lived doesn't mean it's not important. I'd like better control of it for each of the parts on a build during building.

lordofhyphens, 2016-06-26:
You're going to need to come up with a concrete example of *why* you want that kind of control for me to even begin to try to understand it. What problem is it going to solve?

burdickjp, 2016-06-27:
Using PVA support material is a good example. It is very finicky. Having control over extrusion height and width can be the difference between it sticking and it becoming a squid hanging from the extruder.

lordofhyphens, 2016-06-28:
Because it's going to come up, here's the files of interest for adding a new GUI/CLI option. It takes a recompile. 
https://github.com/alexrj/Slic3r/wiki/Code:-Adding-new-option-to-GUI-CLI

my plan was to add an option to replace the assignment in https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/SupportMaterial.pm#L340 with a different option to allow it to be fixed in mm or % of the nozzle width. It's important that it's % of nozzle width to preserve user expectations (e.g. if you leave it at default little changes).

burdickjp, 2016-06-28:
I agree that percentage of nozzle diameter is a good default.
If I'm understanding correctly, the current default is 75% of nozzle diameter? So implementing this feature won't break its current behavior?

lordofhyphens, 2016-06-28:
@burdickjp Correct, 75% is the current default value.

burdickjp, 2016-06-28:
Are you more interested in making it happen yourself, or are you willing to help me through it?

lordofhyphens, 2016-06-28:
either/or, I'm always happy to increase the # of people who are contributing good code to slic3r :)

burdickjp, 2016-06-28:
Definitions, default values, and CLI options all look rather straightforward. I've forked. I'll get those put together.

lordofhyphens, 2016-06-28:
@burdickjp I find it helpful to create a branch in the fork and use just that for the pull request.

We'll also need to come up with some tests for the new functionality. That syntax is more magical to me, but the tests are located in the t/ directory.

burdickjp, 2016-06-29:
So I was looking through the cod and I'm looking at a number of ways this can be accomplished. We could allow the user to define support layer height by any of the following ways:
a number of layers. This is how combining infill works.
a percentage of nozzle diameter. This is how it's doing it now, but is hard-coded
a percentage of layer height. I don't like this one much.
a dimension in mm. This is how Simplify3D works.

We could also allow the user to put in a dimension in mm OR a percentage of nozzle diameter.
Or a number of layers OR a percentage of nozzle diameter.

The way I was thinking earlier, having a number of layers OR a percentage of nozzle diameter would require a new class in xs/src/libslic3r/Config.hpp, and I'm know nothing about C++.
If we did a dimension in mm OR a percentage of nozzle diameter we could use a current class in xs/src/libslic3r/Config.hpp.

Right now I'm leaning toward just a number of layers. Or just a percentage of nozzle diameter.

lordofhyphens, 2016-06-29:
We already have a ConfigOptionFloatOrPercent data type, copy-paste another one of that kind and change the bits to suit what we want. nozzle_diameter would be the ratio_over setting you would be looking for. 
https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/PrintConfig.cpp#L401 for an example.

Number of layers breaks everything unless 1 = current value. If 1 layer = current value, more than 1 is unprintable. So I wouldn't accept it in that case.

burdickjp, 2016-06-30:
What's going on here?
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/SupportMaterial.pm#L44

lordofhyphens, 2016-07-01:
It's calling support_layers_z() with those arguments, which generates all those support layers between the contact and the base.
First argument is all of the contact layers, second is the base layers, third is the thickest object layer height.

burdickjp, 2016-07-01:
Okay. I understand a bit better how that method works.
It seems that passing the thickest object layer height is no longer necessary. I think we should be passing the configured support material layer height. I've made an attempt at it here:
https://github.com/burdickjp/Slic3r/commit/5634e0984017ea503a23ed64d9237aae27ca86f7
but it is definitely not finished and I'm still trying to figure out what I need to do next.