slic3r/Slic3r #3378: Feature request: Thinner outer perimeters

lordofhyphens, 2016-06-26:
Posted by @ldpgh

### Behavior
* _Describe the problem_
    - In order to get a smoother surface (PLA) I'd like to print just the most outer perimeter with a reduced layer thickness (e.g. 0.25mm standard layer height ... just most outer perimeter 0.05mm ... keep everything else as it is)

_Is this a new feature request?_ Yes

#### STL/Config (.ZIP) where problem occurs
_Needed_

lordofhyphens, 2016-06-26:
@ldpgh if you were the one in IRC earlier, the place to go looking around to muck with the perimeters is 
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/Object.pm around sub make_perimeters. combine_infill is in the same file. 

So long as you can build/run Slic3r all you have to do is edit that file and restart Slic3r for the changes to take effect (it doesn't take a rebuild to effect Perl changes).

Adding it as a configurable option in the GUI is a bit more involved, so I'd suggest testing with a fixed multiple first. If the results look promising I can help you get the configuration stuff going.

ldpgh, 2016-06-27:
Thx for adding this issue.
Here come IRC-screenshots made to illustrate a feature request and the simplified cross section Mikk36 has drawn.

ldpgh, 2016-06-27:
Nice URL .... http://richrap.blogspot.de/2012/01/slic3r-is-nicer-part-3-how-low-can-you.html ....  need to keep it in mind when checking the results.
My test of 0.25mm vs. 0.05mm layer thickness for an 10x10x5mm^3 cube (PLA) looks similar.

ldpgh, 2016-06-27:
My Slic3r installation (Rev 1.0.0RC3) does not contain any Perl-files (part of the K8200 setup) ... just DLL files found.

ldpgh, 2016-06-27:
Spend whole day to get an installation running ... but dmake.exe & C-compile is not working properly and rejects installation of required modules (e.g. XS).

Need to stop at this point to not damage my existing C/C++ installation.

lordofhyphens, 2016-06-27:
Download a portable version of Strawberry Perl and run the build commands
from that. It should work w/o any issues.

Slic3r 1.0RC3 is ancient; use the wiki instructions
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows to
produce a new build that you can edit the perl sources of.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

ldpgh, 2016-06-27:
Thx ... following  your recommendation ... any hint for IDE w/ debugger included.

lordofhyphens, 2016-06-27:
Nope, not on Windows. I don't do any dev work on Windows and my knowledge
of Perl IDEs is thin on the ground (even if they work). I don't use an IDE
on Linux either, my usual text editor is good enough for anything I want to
use.

Perl is interpreted, so if you're editing a .pm file, just liberally seed
the area you are interested in with print statements. Restarting Slic3r is
sufficient to see your changes.

Just keep the perl window open and run slic3r with "perl slic3r.pl" (from
the strawberry perl command line) to see the console output.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2016-06-27:
Regarding debugging on Windows:

It is indeed possible to debug Perl and it is extremely helpful for a Perl newbie. The EPIC plugin for Eclipse works very well with an older eclipse-SDK-3.6.2. There is a catch though: The Perl debugger does not work correctly with multiple threads running under the Perl interpreter. If that happens, the EPIC plugin gets confused and the debugger stops working. The same happens with the Komodo IDE, the commercial alternative.

Debugging the C++ code works fine using the latest Eclipse for C++ and the gdb of MinGW. The gdb packed with the Strawberry distribution does not contain the Python support, so pretty printing of the stl containers only works if another gdb build is used.

ldpgh, 2016-06-27:
Help really appreciated ... while still running WinXP   :-o

Thought 1st step is "perl Build.PL"
```
>>> perl Build.PL
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
....
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
Don't worry, this module is optional.
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
...
Can't write to cpanm home 'C:\DOKUME~1\.\E.IGENE~1/.cpanm': You should fix it with chown/chmod first.
Don't worry, this module is optional.
Can't write to cpanm home 'C:\DOKUME~1\...EIGENE~1/.cpanm': You should fix it with chown/chmod first.
...
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
Don't worry, this module is optional.
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
Can't write to cpanm home 'C:\DOKUME~1\...\EIGENE~1/.cpanm': You should fix it with chown/chmod first.
The XS/C++ code failed to compile, aborting`
>>>
```
```
>>> perl -v
This is perl 5, version 24, subversion 0 (v5.24.0) built for MSWin32-x86-multi-thread-64int
Copyright 1987-2016, Larry Wall
>>>
```
```
>>> perl slic3r.pl
Can't locate Slic3r/XS.pm in @INC (you may need to install the Slic3r::XS module) (@INC contains: C:/....../Slic3r/lib C:/Perl/Strawberry/perl-5.24
.0.1/perl/site/lib C:/Perl/Strawberry/perl-5.24.0.1/perl/vendor/lib C:/Perl/Strawberry/perl-5.24.0.1/perl/lib .) at C:/...../Slic3r/lib/Slic3r.pm
line 37.
BEGIN failed--compilation aborted at C:/...../Slic3r/lib/Slic3r.pm line 37.
Compilation failed in require at slic3r.pl line 15.
BEGIN failed--compilation aborted at slic3r.pl line 15.
>>>
```

lordofhyphens, 2016-06-27:
see bottom of
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-06-27:
Also, make sure you can replicate your problem/failure case (what prompted you wanting to change it) on current latest master before changing anything :)

ldpgh, 2016-06-27:
1. Last comment not clear to me ... have not changed anything yet.
2. Why  XS.pm  is not loaded properly
3. Why does a property change of a non-existing directories can prevent the "Can't write" warning? Btw. .cpanm is created in my XP home directory "C:\.......\Dokumente und Einstellungen" (with "spaces" !) rather than in "C:\".

lordofhyphens, 2016-06-27:
1) Meant just make sure that current version of slic3r (1.3.0-dev) doesn't
already do what you are interested in changing... a lot has changed since
1.0RC3
2) Don't have an answer for that.
3) I have not delved into the mysteries of Windows XP in many moons, I
cannot help you there.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

ldpgh, 2016-06-28:
1. `Can't write to cpanm home 'C:\DOKUME~1\...EIGENE~1/.cpanm': You should fix it with chown/chmod first.` seems to be more a real issue ... at least for my WinXP. 
.
Statements added to control home directory naming and location ...
`C:\....\Slic3r >>> set PERL_CPANM_HOME=C:\CPAN_HOME`
`C:\....\Slic3r >>> mkdir %PERL_CPAN_HOME%`
`C:\....\Slic3r >>> perl Build.PL`
`--> Working on App::cpanminus`
`Fetching http://www.cpan.org/authors/id/M/MI/MIYAGAWA/App-cpanminus-1.7042.tar.gz ... OK`
`Configuring App-cpanminus-1.7042 ... OK`
`Building and testing App-cpanminus-1.7042 ... OK`
`Successfully installed App-cpanminus-1.7042 (upgraded from 1.7041)`
`1 distribution installed`
`....`
.
and the output looks much more promising. Not clear whether the spaces in the directory name or the '~' in the shorted directory name are the root cause. Now Just boost needs to be added.

2. (note) `perl Build.pl` does not list the missing prerequisites
   - it issues a `die`-statement due to XS/C++ issues right before this information is printed
   - the real issue (modules not loaded) as solved in 1. is hard to detect.

3. Next step boost_1_61_0 ...
`bootstrap mingw` seems to become  `bootstrap.bat gcc` ... to solve `### Unknown toolset: mingw`

ldpgh, 2016-06-28:
1st step ... comparing results by program revision (same input and conf-file used)

; generated by Slic3r 1.3.0-dev   (new setup)
- active_lines=601 of 742
- filament=25.9mm
- distance (left <-> right) of inner perimeter center line = larger

; generated by Slic3r 1.0.0RC3 (current setup)
- active_lines=467 of 487
- filament=21.3mm
- distance (left <-> right) of inner perimeter center line = my reference

lordofhyphens, 2016-12-19:
@bubnikv has a decent working debugger setup going on Windows by the way. (also tagging to bring his attention to this if he wants to take a crack at it).

alexrj, 2016-12-19:
This feature request lacks a proper description of how the feature is supposed to work. It looks only applicable to vertical walls. Also, it lacks a discussion of what advantages would this result have over the already existing "Combine infill" feature.

lordofhyphens, 2016-12-19:
@alexrj yeah, I had it marked as PR or bust for a reason ;)

I'm willing to look over code from @ldpgh if <insert pronoun here> is willing to work on it, but I wasn't really convinced of the utility over combine infill (the only real savings I could see was some time savings on perimeters myself). My assumption is that @ldpgh would be able to produce a convincing argument for its inclusion if <insert pronoun here> was able to work out a decent method and describe it. If not, then it needn't be merged. 

If you want to tag as "Won't change" and close as an indication that you probably wouldn't merge it than go right ahead and let @ldpgh figure out a better description for it. I'm the originating author because this was split off from an IRC conversation, if memory serves.