slic3r/Slic3r #3382: Wishlist: Modify extrusion multiplier only for first layer

KlausKnopper, 2016-06-28:
### Version
1.3.0-dev

### Operating system type + version
Debian GNU/Linux stretch

###  New feature request:

Especially for ABS filament, printing the first layer with excessive (i.e. about 125%) extrusion, thus causing higher pressure of material on the print bed, may result in better adhesion and less warping. This is apparently not the same as increasing layer height or extrusion width, which can already be changed for the first layer, since these settings will in relation change the nozzle distance of the neighbour paths as well, whereas an extrusion multiplicator will actually purpously cause overlaps and additional glue.

This new feature probably fits best in the "filament" profile, since it may only be especially useful for ABS-type filaments that shrink quickly and therefore are critical for first layer adhesion.

whosawhatsis, 2016-06-28:
This has been requested and denied several times. If you want a larger extrusion on the first layer, use the first layer height and extrusion width settings. If you want to overpack the first layer, adjust your printer so that the first layer prints with the nozzle closer to the platform. We used an extrusion multiplier for this in the bad old days of slicers to effectively increase the first layer height, but there are better ways to do these things now.

mrvn, 2016-06-30:
Or define a Z offset for the print bed.

lordofhyphens, 2016-06-30:
This may be solved through the clever application of #3390 when I get it sorted out, as I think the current layer # is exposed as a variable (or at least the current Z); you could set the extrusion multiplier and the use this conditional gcode.

Another workaround is to set your extrusion multiplier higher to get what you want and then add a M221 command in the after_layer_change custom gcode to reset the flow amount. Since the value isn't changing on subsequent layers it should be safe. It'd take some math to work out the combination of extrusion_multiplier and reduction to get it right, but it's certainly serviceable.

surdu, 2017-04-12:
Customizable extrusion multipliers would also be nice for different infills, I guess. My last layers can hardly cope covering even 20% infills. 

Might also be solvable using extrusion width, trying as we speak, but I have low hopes ...

lordofhyphens, 2017-04-12:
@surdu there is something else wrong with your printer if you are trying to change the extrusion multiplier like thst.

surdu, 2017-04-12:
This is how it looks after 2 layers of top solid infill: http://imgur.com/a/gnIzm

 Looks to me like not enough plastic is extruded ...

lordofhyphens, 2017-04-12:
@surdu the first layer of top solid is done with bridging flow. If your printer can't manage those bridges... the rest of the layers will fail (unless you put enough on there that it is hidden)

Also, ensure you are using 1.3.0-dev.

There may be a problem with how much flow (that would be separate issue and should be written up as much).

Hacking the extrusion mult is not the correct solution for your problem.

hexane360, 2018-04-23:
@whosawhatsis @mrvn Simply adjusting Z-offset doesn't provide enough DOF to calibrate a printer, because Z-offset is also necessary to adjust dimensional accuracy on the Z-axis.

Just like bridging is a special case of flow math, so is printing on a bed. It should have modified flow math or the option for users to manually modify its flow math.

l29ah, 2018-06-20:
Duplicate of #3013?