slic3r/Slic3r #3383: NOTICE: -- Slic3r's general TODO list --

lordofhyphens, 2016-06-28:
https://github.com/alexrj/Slic3r/wiki/TODO

Wanted to make this a bit more visible for the different ways you can help solve problems (including documentation/UI) with Slic3r.

Patola, 2016-10-07:
This todo list is too short and weird. Dual extruders work in slic3r for a long time already, for example. And better support might be resolved by your pull request with more design patterns, turning on/off the support sheath and support only from bed feature. This way a person who goes there is kind of stumped on what exactly to do... The "need advice and ideas from community" issues are much better in this respect.

lordofhyphens, 2016-10-07:
I have been trying to categorize things; suspect that the todo list is best
summarized with a link to the pr or bust milestone or the individual
filters (particularly the feature request tag).

alexrj, 2016-12-18:
Oh, I just noticed that page - it's very out-of-date, maybe better delete it... no?

lordofhyphens, 2016-12-18:
That or restrict it down to what I suggested, use labels and/or milestones
to implement our wish list

andy5995, 2019-01-10:
I've just submitted [a related PR](https://github.com/slic3r/Slic3r/pull/4690).

I noticed that some information is duplicated on CONTRIBUTING.md and the TODO.

A separate ticket might be needed so the information can get reviewed more in-depth, and the pages get merged.