slic3r/Slic3r #3388: Bad bridging direction in git master

thaytan, 2016-06-30:
### Version
git master - commit #cb7296 (1.2.9-266-g259eb99)
on Fedora 24

I'm slicing a piece, and slic3r is picking the longest bridging direction. In 1.2.9, it bridged orthogonally to that, picking the shortest direction. Actually, in 1.2.9 the bridging direction depends somewhat on part orientation on the bed - rotating the part produces incorrect bridging. In git master, it's incorrect in any orientation I tried.

The full model is quite big (22MB zipped), but I can upload if wanted. I made a cut down version around the problem area, and it behaves slightly different in that the bridging direction is now also wrong in 1.2.9 in any orientation, which is interesting.

thaytan, 2016-06-30:
github is refusing to upload any of my files to this bug - I'll try again later.

lordofhyphens, 2016-06-30:
:(
https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/
Might be case-sensitive on the extension.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

thaytan, 2016-07-01:
[image]
[image]

There's the images. It's still refusing to let me attach my zip file with config and STL, so find it here: http://noraisin.net/down/top-trimmed.zip

thaytan, 2016-07-14:
I implemented a new method of choosing bridging angles. It gives some surprising results - small variations in the 'optimal angle' depend on nozzle diameter / extrusion width, but otherwise seems fairly stable, and chooses good directions. Part orientation can sometimes affect the result slightly, but only small changes in angle that are within the 'noise' based on slic3r coordinates resolution.

Lenbok, 2016-07-31:
How does your new method do on the case that nophead discussed in  http://hydraraptor.blogspot.co.nz/2016/01/a-bridge-too-far.html

thaytan, 2016-07-31:
It does the same as unmodified git - aligns the bridging "optimally" perpendicularly across the gap, but I notice that if I add more perimeters, then both git and my direction metric produce the same output, bridging at 45 degrees with nothing underneath. 
[image]

It's not surprising that both git master and my direction metric have the same problem though  - I didn't change anything about logic that's supposed to reject invalid candidate directions.

thaytan, 2016-07-31:
Here it is with only 1 perimeter:
[image]

alexrj, 2017-05-16:
This was fixed in 1.3.0-dev. Get it at http://dl.slic3r.org/dev/
Thank you for the report! Screenshot follows:

[image]

closed 2017-05-16 by alexrj

thaytan, 2017-05-21:
Great! Confirming - the new bridging calculation works much better.