slic3r/Slic3r #3389: Tool change g-code for independent extruder lifting (dual extruders)

jaredabrandt001, 2016-06-30:
### Version
Slic3r 1.2.9

### Operating system type + version
Windows 7
Marlin RC6 firmware on RAMPS1.4

### Behavior
This machine has dual extruders with a servo to retract the 2nd extruder (T0) independently.  I can include M280 to activate the extruder, but I do not understand how to include a conditional statement so that the servo lowers when T1 is active and raises when T0 is active.  Is there a way to do this in the tool change g-code box?  If not, do you suppose there's a way to program this into Marlin and activate via tool change g-code box?

Thanks in advance!
Jared

lordofhyphens, 2016-06-30:
Conditional statements are not currently a thing in gcode; best immediate solution I can give is to hack together a short script that looks for the toolchange gcode (slic3r should insert a T0 or T1 to tell the firmware to use that second extruder) and implement your condition there; you'd need to insert a new line or replace an existing one. 

It'd probably be easier to insert a line with the toolchange gcode with a reference to the current extruder in a comment and do a find/replace on that to work out what to do. 
reference: https://github.com/alexrj/Slic3r/wiki/FAQ https://github.com/alexrj/Slic3r/wiki/Writing-post-processing-scripts

so in the toolchange gcode you'd do something like 
`; next: [next_extruder]`

and write a script that looks something like the below script. 
````
#!/usr/bin/perl -i                                                     
use strict;
use warnings;

while (<>) {
     if ($_ == "; next: T0") {
          $_ = "gcode to lower servo goes here";
     }
     if ($_ == "; next: T1") {
          $_ = "gcode to raise servo goes here";
     }
     print;
}
````

lordofhyphens, 2016-06-30:
I may think of a usable syntax for a generic post-processor for custom gcode so people can have conditional statements; etc in them. I think it'd start limited and it'd probably belong in another github issue.

lordofhyphens, 2016-06-30:
The generic part of your request is covered by #3390 and I answered how to make this work for you before that goes in, so I'm going to close this issue now. Feel free to add comments here if you have further questions.

closed 2016-06-30 by lordofhyphens

jaredabrandt001, 2016-06-30:
Wow, thanks...that was fast!
Yea I read that scripts and/or post processing/ using find and replace could be the answer.  This might get me by for now, and thanks!
The ultimate goal for this printer is to be business-class circuit board printer, and slic3r is the only slicer that processes the traces correctly at this time.  To this end, I think I will try to hack into the Marlin T0 and T1 routines to see if I can get the firmware moving the servo for me.  I'm not much of a programmer so I'll have my fingers crossed:)

lordofhyphens, 2016-06-30:
The solution I outlined should work for you (better than hacking Marlin and maintaining that hack), but keep an eye on #3390 and that should give a much more flexible solution.

lordofhyphens, 2016-07-01:
https://github.com/lordofhyphens/Slic3r/blob/conditional-gcode/utils/slic3r-conditional.pl 
Here's a script that implements conditional inclusion of gcode as a post-processor.

It's using eval() so it needs to be a legal Perl expression, so == instead of =. Depending on how crazy you get re: perl you could do some gnarly things with this as it is.