slic3r/Slic3r #3393: Bridge issues

JuTakata, 2016-07-03:
### Version
1.2.9

### Operating system type + version
Windows 7

### Behavior

Hi,

I am not sure if this is due to my configuration or if this is a bug but I when I try to slice the ozzebane_test calibration I get gaps on my print beetween top solid infill and "towers perimeters".

This problem appear when I use 0% infill as you can see on the attached screenshots there is no infill under tower perimeters and the 45° bridge/top solid infill. 
This bridge should not be perpendicular to the part instead of 45°? It seems like if there was no bridge but only solid infill.

First layer that should be bridge :
[image]
First tower perimeters :
[image]

With 5% infill the bridge is also printed at 45° but there is a top solid infill under tower perimeters and then there is no problems :

[image]
[image]

Thank you for helping me to understand if this is a configuration issue or a bug.

#### STL/Config (.ZIP) where problem occurs
[Slic3r issues.zip](https://github.com/alexrj/Slic3r/files/344976/Slic3r.issues.zip)

PS : Sorry for my not perfect English, if something is not clear please feel free to ask for details

lordofhyphens, 2016-08-24:
No infill, no bridging infill, so far as I can tell.

alexrj, 2017-05-16:
@JuTakata, you need infill for that. There are no anchors for a bridge. That object can't be printed at 0%.

closed 2017-05-16 by alexrj