slic3r/Slic3r #3397: Feature Request: Configureable distance between object top layer and support material bottom

Ricky1975, 2016-07-04:
### Version
1.2.9-276-g8a72ffb

### Operating system type + version
Any

### Behavior
When using support inside an object (lower end of the support is touching a surface of the object), the distance between object and support seems to be the normal support layer height. This leads into sticky support.
I would expect to have it configurable manually like the "support contact z-distance" for the top interface or at least reuse this variable.
When I looked at the SupportMaterial.pm I assume, there is currently no explicit handling of the bottom distance and it's handled by the object clipping routine.

lordofhyphens, 2016-07-05:
Yeah, that's how it's done currently; not a bad idea.

lordofhyphens, 2016-07-05:
Related to #3081