slic3r/Slic3r #3398: Allow to specify default extruder

DrLex0, 2016-07-05:
### Version
1.2.9

### Request
There should be a way to set the global default extruder, preferably in the Printer Settings, but maybe also in a drop-down menu in the Plater tab (see issue 2070). An item in the Printer Settings would be useful because I already have settings for “left only,” “right only,” and “both,” each with custom code for wiping the nozzles and doing the tool change in a way that does not confuse my FlashForge Creator Pro.
When I select my “left only” profile, the default nozzle should become T1 instead of T0, otherwise I explicitly need to specify the nozzle for the object. If I don't do that (and I obviously will forget it from time to time), certain operations will be performed with the wrong nozzle, for instance when setting a different temperature for the first layer, T0 would heat up after the first layer, instead of T1.

alexrj, 2016-07-10:
Why don't you just add the string `T1` in the start G-code of your "left only" profile?

DrLex0, 2016-07-10:
I did, but that doesn't help. If the object is _not_ explicitly set to use extruder 2, and I have configured a different filament temperature for the first layer, then T1 will not switch to the second layer temperature. If I have selected the same filament for T0 in the Plater settings, then T0 will heat up to the temperature T1 should be heating up.
The best I could come up with, is adding this line of G-Code while the printer waits for the nozzles and bed to heat up…

M70 P5; Forgot to set nozzle2 for object?  ==>  Cancel now and fix!

This shows up on the 20x3 display as:
Forgot to set nozzle
2 for object? ==>
Cancel now and fix!

Better than having to scrape off the first layer and starting over :)

DrLex0, 2016-07-10:
You don't need to start a print on a dual extruder setup to see the problem. Here's a recipe (I should have provided this from the start, given that it took me about an hour to figure out what was going wrong):

1. Use the attached config: [Slic3r-Dual-LeftNozzle.ini.txt](https://github.com/alexrj/Slic3r/files/356182/Slic3r-Dual-LeftNozzle.ini.txt)
2. Create a filament with different temperatures for the first layer and other layers.
3. In Plater, select the imported profile for everything (use the same filament for both extruders).
4. Load any object and export the G-Code.
5. Look into the generated code, you will find a line "M104 Sxxx T0 ; set temperature" after the code for the first layer.

If you go into the object's settings and set its extruder to 2, then the command will use T1.

This is not really a bug because Slic3r has no way of knowing that it should be heating T1 in this scenario. But if the default extruder could be set together with the printer profile, then there would be no need to manually set the extruder for the object.

freiser77, 2017-03-04:
Another way would be to choose the tool index in the settings of the printer.
This would be very convenient together with being able to name the extruders other than "Extruder 1", "Extruder 2", etc.

DrLex0, 2017-03-04:
Yes, what we really need is a way to indicate what the ‘main extruder’ for a printer profile is. When the profile specifies the second extruder as main extruder, the bed temperature will be set according to that extruder. I'll try to rename this issue to reflect this.

DrLex0, 2017-06-02:
I'm going to close this one because when setting up everything the right way (single extruder, T1 command in start G-code etc), there is only one real issue left, which is that Slic3r still appends an incorrect T0 argument to the command to change temperature after the first layer.

closed 2017-06-02 by DrLex0

lordofhyphens, 2017-06-02:
@DrLex0 is that "one real issue left" captured somewhere else?

DrLex0, 2017-06-02:
Yes, I just created #4003