slic3r/Slic3r #3399: Fermats spirals

kengdahl, 2016-07-06:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

This is a feature requset. Have you thought about implementing fermats spirals as a infill method?
Here is a paper on the subject: https://www.cs.sfu.ca/~haoz/pubs/zhao_sig16_fermat.pdf

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

mrvn, 2016-07-06:
I really like the idea although the pictures in the paper don't show such a good result. The zig-zag actually looks better. But how much of that is the printer? Or over/under filling mentioned in the paper.

What I really is that it constructs a single continious loop, assuming there is only one island.

- no retraction
- no crossing perimeters
- easy to not start/stop on an outer perimeter (not done in the paper it seems)
- assuming all layers have only a single island one could even print the whole objects in one continious tool path.

I don't think this will be good for infill though. Path placement depends on the overall shape so it will differ from layer to layer and not garantee the current layers path will rest on the previous layers infill. Also, similar to concentric, it lacks stability. The infill needs a crossing pattern to give stability in more than one direction.

I think this would be more suitable for top/bottom fill and solid fill.

lordofhyphens, 2016-07-07:
If someone wants to prototype it and submit it as a PR with print tests showing it's worth having, then I'd probably merge it in.

Zule10, 2017-04-26:
Good afternoon,

I would like to work in the fermats spirals code for its implementation, though I don’t have enough experience programming for Slic3r. What I mean is that I want to try to solve the challenge, but I need help doing the transition or the shift between my code in C++ and the one in which Slic3r works or is programmed. More documentation or help about this challenge is openly welcomed.

Zuleimi Gonzalez, developer, currently studying systems (software) engineering at Universidad EIA, Colombia.
You may leave us a comment or contact us in @Zule10 @felohdez2008

lordofhyphens, 2017-04-26:
Slic3r's infill is in C++.

jcongote, 2017-04-27:
@lordofhyphens I am John Congote, and I will help @Zule10 and @felohdez2008 to develop the infill algorithm, however if you can point out a function in the source code that makes an infill pattern could be helpful so we can check the input of the function and the expected output... and start implementing the fermat spiral under the constrains of the sourcecode.

lordofhyphens, 2017-04-27:
https://github.com/alexrj/Slic3r/tree/master/xs/src/libslic3r/Fill

Go nuts.