slic3r/Slic3r #3401: Feature Request: Z-Hop Time-Saving Feature

rf1k-mjh11, 2016-07-07:
### Version
1.29 - and previous versions

### Operating system type + version
Win 7  -and any other operating system that supports Slic3r

### Behavior
New Feature Request (modification of existing feature)

**Current Condition:**
The current functionality of Z-Hop requires that the nozzle be raised **EVERY TIME** a retraction is performed.
Z-Hop appears to be (in my opinion) intended as an appearance improvement by reducing the marks left by the nozzle as it crosses over areas with previously deposited filament.

**Suggested Modification** to Z-Hop:
Z-Hop should only be performed when crossing a '_top surface_' layer.  Where _top surface_ layers are defined as those parts of a layer that do not have a layer immediately above them.

Benefits:
- Significant time savings while maintaining appearance expectations.  
- Any visible area has a good chance of not having visible marks.  
- No Z-Hops are performed in layers that are not visible in the end product.

### Unnecessary Comments
I've never used Z-Hop, but the idea came up due to a forum post complaining about the marks made by the nozzle when crossing filled areas.
With the time penalty of Z-Hop largely negated, maybe more people would use it.

mjh11

mrvn, 2016-07-07:
Don't forget perimeters, they are visible too. Z-hops reduce zits and strings and most importantly collision of the head with the object when crossing over a hole. You need this on delta printers because Z flutters as you move in XY and when corners curl up (overhangs).

Z-hop may makes sense for internal spaces too since it reduces drag on the object preventing it detaching and getting pushed around the plate. Might be negible.

Note: When moving from one island to another (crossing perimeters) instead of moving straight up and down in Z slic3r could generate moves that change Z while moving the head so that it reaches the hop height before crossing the first perimeter and maintains it till after it crossed the last perimeter. This could also reduce artefacts from the lift itself since it would be a wipeing motion.

Fun fact: Z-hop is also essential when abusing slic3r to generate g-code for a cnc mill. Otherwise you would cut through material that is supposed to remain.

lordofhyphens, 2016-07-11:
You also don't need much movement, a layer's worth of movement is sufficient unless you're trying to dodge curling (but this also seems doomed to failure as you're going to be moving over that area anyway).

mrvn, 2016-07-12:
With curling the material is usually thin and still flexible. When moving from the inside to outside the nozzle just pushes it down. But when crossing the perimeter from the outside in it will push the curled edge further up and break the part loose. So to fight curling a z-hop of a few mm is not out of the question.

VanessaE, 2017-05-09:
*bump* 

@mrvn's comment from 2016-07-07 would be rather useful.  "Avoid crossing perimeters" works for simpler models, but for anything particularly complex, it just makes slicing way too slow.  On the other hand, it should be trivial to do a Z-lift when (and *only* when) crossing a perimeter or a top-solid surface, instead of trying to find an alternate path.