slic3r/Slic3r #3405: Failing tests in 01_trianglemesh related to new cppsvg branch

lordofhyphens, 2016-07-10:
The split and merge tests appear to fail because they're not doing anything (split is returning half is expected, merge is returning double, etc).

Test output:
   Failed test 'merge'
   at t/01_trianglemesh.t line 70.
          got: '12'
     expected: '24'

   Failed test 'split'
   at t/01_trianglemesh.t line 74.
          got: '1'
     expected: '2'
 Looks like you failed 2 tests of 49.

Did a git bisect on the cppsvg branch, looks like the first one that fails tests is 
7f2e774584a59b331543265971fdd1bb796de929

Had worked before the merge occurred, so it looks like something went awry when the TClap dependency was removed. That commit touches a lot, which makes it a pain to revert. I'm looking into what it changed to see what could have gone wrong, but it's a bit of fumbling in the dark :(

7f2e774584a59b331543265971fdd1bb796de929 is the first bad commit
commit 7f2e774584a59b331543265971fdd1bb796de929
Author: Alessandro Ranellucci <aar@cpan.org>
Date:   Sun Jul 3 10:15:46 2016 +0200

    Removed dependency on TClap

:040000 040000 9ff011f84b178f46d9866a79128e9e5325b6b4d3 701b6e7ea479992d353a755b6c6af3fbdac2d73a M	src
:040000 040000 12d6f8bfd3e94f247a0bda52df5eb10fe383835b f3bc77f4d63667125f76e3236aae09c0b0f87d22 M	xs

lordofhyphens, 2016-07-10:
I'm wondering if it's the change to swap right now. https://github.com/alexrj/Slic3r/commit/7f2e774584a59b331543265971fdd1bb796de929#diff-c0c682ae79ed7d528e970ec16489e2b4R58

alexrj, 2016-07-10:
Fixed, thank you.

closed 2016-07-10 by alexrj